armbian/build #314: S20armhwinfo no such file erros on SinoVoip/M2+

Grabber, 2016-05-22:
[image]

damien7851, 2016-05-23:
Hello, you probably off the green led in your fex file.
what kernel you use?

Grabber, 2016-05-23:
Igor,

I compiled my own kernel version using lib project from your github
account, just enabled bonding networking as a module.

Best,
Luiz.

ThomasKaiser, 2016-05-24:
> you probably off the green led in your fex file.

There is no green led on BPi M2+ (but since it took the vendor a few weeks and countless reminders to correct the board's 'specifications' this is what can happen then). 

It's safe to ignore the output above. In case you feel this should be corrected feel free to send a pull request redirecting `stderr` to `/dev/null` from within `armhwinfo` (but you should keep in mind that we're aware of the led issues and try to fix this with one of the next releases anyway -- ATM I would believe some of us are waiting for 5.11 release while Igor is waiting for some of us commiting stuff ;) )

Anyway: This is not a real issue and might be closed right now.

igorpecovnik, 2016-05-24:
5.11 is ready but since u-boot for most allwinners was changed to recent 2016.05... it would be nice to do some tests before putting out. I am working to establish test repository, which can be populated with untested stuff ... later today.

Grabber, 2016-05-24:
I can help you guys test anything... just tell me what and how.

damien7851, 2016-05-25:
I watch "armhwinfo" and I find that the green LED is used before detecting the type of card. should therefore put detectection ahead and add a test to not use the LED on cards that do not.
that's a pretty significant change that simple. And I think it is better to let igor make choices. especially that this bug is not annoying. Igor would be but can have a pull request ...
In this case I could possibly make this change but I could not test it

Grabber, 2016-05-25:
@damien7851 I can provide remote access to M2+ or help test if you provide the patch + steps.

ThomasKaiser, 2016-05-25:
We decided already to drop the whole auto detection stuff since it can not work reliably now that clone after clone appears (BPi M2+ being just one of them). In case you feel offended by startup messages feel free to add to every occurence of `trigger` a `2>/dev/null` in `/etc/init.d/armhwinfo`

ThomasKaiser, 2016-06-05:
Since this was fixed as part of #316 and we're again reworking led behaviour issue is closed now. Please report back if you get similar issues with 5.15 or above.

closed 2016-06-05 by ThomasKaiser