armbian/build #315: [OPi PC] CPU detected correctly in 4.6 kernel

kometchtech, 2016-05-24:
Verify kernel 4.6 with dev on test construction Armbian where the CPU is not recognized correctly.
```
root@orangepipc:~# lscpu
Architecture:          armv7l
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0
Off-line CPU(s) list:  1-3
Thread(s) per core:    1
Core(s) per socket:    1
Socket(s):             1
Model name:            ARMv7 Processor rev 5 (v7l)
```
http://forum.armbian.com/index.php/topic/1242-opi-pc-cpu-detected-correctly-in-46-kernel/

ThomasKaiser, 2016-06-02:
Please read carefully through [my answer](http://forum.armbian.com/index.php/topic/1242-opi-pc-cpu-detected-correctly-in-46-kernel/?view=getlastpost) and then close this (non-)issue :)

kometchtech, 2016-06-02:
Let me confirm forum.

closed 2016-06-02 by kometchtech