armbian/build #318: Build failure for orangepione in Docker

pietrushnic, 2016-05-25:
I'm trying to compile Orange Pi One image in Docker (as described in #255) and facing couple problems. My host is Debian Sid. Configuration:

```
./compile.sh BOARD=orangepione PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=default CLEAN_LEVEL=cache
```

[terminal log](https://gist.github.com/pietrushnic/c909a0ebecc4267ff8177c306c07346a)

This lead to incorrect URL - `E: Failed getting release file http://localhost:3142/httpredir.debian.org/debian/dists/jessie/Release`. Looks like this was caused by incorrect result in `lib/debootstrap-ng.sh:133`. I tried to use `NO_APT_CACHE=yes` but it leads to other problems.

For `EXTENDED_DEBOOTSTRAP=no` I'm getting out of loop devices in container.

Any ideas what I can do to build OPi One image ?

zador-blood-stained, 2016-05-25:
> NO_APT_CACHE=yes

It should be `NO_APT_CACHER=yes`, and it is required if you are building inside a container when your host OS doesn't have apt-cacher-ng running.

pietrushnic, 2016-05-25:
How I can check if `apt-cacher-ng` is running on host ?

zador-blood-stained, 2016-05-25:
It isn't installed by default - so unless you install it explicitly, it isn't there. You can check if it is running by opening `http://<ip-address-of-your-build-host>:3142` in web browser.

pietrushnic, 2016-05-25:
It looks like I have it installed on the host. Entering my `<ip>:3142` into browser give me:

```
You attempted to browse the contents of a virtual HTTP repository. However, the intended way of use is the configuration of APT and related package management systems to retrieve the software packages through this service.
(...)
```

pietrushnic, 2016-05-27:
@zador-blood-stained I retested with recent code 3305dd45be05 using 

```
./compile.sh BOARD=orangepione PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=default
```

I have `apt-chacher-ng` installed and working, but still getting error:

```
[ o.k. ] Creating board support package [ orangepione ]
[ o.k. ] Fingerprinting [ Armbian 5.12 Orangepione Debian jessie default Linux ]
[ o.k. ] Building package [ linux-jessie-root-orangepione ]
[ o.k. ] Starting build process for [ orangepione jessie ]
[ o.k. ] Creating new rootfs for [ jessie ]
[ o.k. ] Installing base system [ Stage 1/2 ]
I: Retrieving InRelease 
I: Failed to retrieve InRelease
I: Retrieving Release 
E: Failed getting release file http://localhost:3142/httpredir.debian.org/debian/dists/jessie/Release
cp: cannot create regular file '/root/output/cache/sdcard/usr/bin/': No such file or directory
[ o.k. ] Installing base system [ Stage 2/2 ]
chroot: failed to run command '/bin/bash': No such file or directory
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:162 ]
[ error ] Debootstrap base system second stage failed #
[ o.k. ] Process terminated #
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:527 ]
[ error ] debootstrap-ng was interrupted #
[ o.k. ] Process terminated #
```

Do I have to redirect ports or something ?

zador-blood-stained, 2016-05-27:
I have no idea how Docker shares network between host system and containers by default, so you may need to either set `NO_APT_CACHER=yes` for further testing or set `APT_PROXY_ADDR` to local IP address of your build host.

pietrushnic, 2016-05-27:
For `NO_APT_CACHER=yes` I get:

```
[ o.k. ] Creating board support package [ orangepione ]
[ o.k. ] Fingerprinting [ Armbian 5.12 Orangepione Debian jessie default Linux ]
[ o.k. ] Building package [ linux-jessie-root-orangepione ]
[ o.k. ] Starting build process for [ orangepione jessie ]
[ o.k. ] Creating new rootfs for [ jessie ]
[ o.k. ] Installing base system [ Stage 1/2 ]
I: Retrieving InRelease 
E: unknown location httpredir.debian.org/debian/dists/jessie/InRelease
cp: cannot create regular file '/root/output/cache/sdcard/usr/bin/': No such file or directory
[ o.k. ] Installing base system [ Stage 2/2 ]
chroot: failed to run command '/bin/bash': No such file or directory
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:162 ]
[ error ] Debootstrap base system second stage failed #
[ o.k. ] Process terminated #
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:540 ]
[ error ] debootstrap-ng was interrupted #
[ o.k. ] Process terminated #
root@57f5a2322b47:~# 
```

This looks like substitution in bash doesn't seem to work as expected.

zador-blood-stained, 2016-05-27:
OK. This will be fixed, meanwhile you can try to apply [this](https://github.com/zador-blood-stained/lib/commit/18509159cbbdf02c9bdea3a81423073693244678) change manually.

pietrushnic, 2016-05-28:
@zador-blood-stained this fix unfortunately do not improve situation, what is needed is something like:

```
 diff --git a/debootstrap-ng.sh b/debootstrap-ng.sh
index 36600a3..5bfbd59 100644
--- a/debootstrap-ng.sh
+++ b/debootstrap-ng.sh
@@ -131,7 +131,7 @@ create_rootfs_cache()
                if [[ $NO_APT_CACHER != yes ]]; then
                        # apt-cacher-ng apt-get proxy parameter
                        local apt_extra='-o Acquire::http::Proxy="http://${APT_PROXY_ADDR:-localhost:3142}"'
-                       local apt_mirror="http://${APT_PROXY_ADDR:-localhost:3142/}$APT_MIRROR"
+                       local apt_mirror="http://${APT_PROXY_ADDR:-localhost:3142/}/$APT_MIRROR"
                else
                        local apt_mirror="http://$APT_MIRROR"
                fi
```

Or some note in documentation that `APT_PROXY_ADDR` should be provided like `192.168.0.107:3142/`. Second thing to mention is that port have to be provided, without that build process do not pass `Installing base system` phase.

Above code improve situation, but `Updating package list` seem to not use proxy:

```
[ o.k. ] Updating package list [ jessie ]
Err http://httpredir.debian.org jessie InRelease
  
Err http://httpredir.debian.org jessie-updates InRelease
  
Err http://httpredir.debian.org jessie Release.gpg
  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]
Err http://security.debian.org jessie/updates InRelease
  
Err http://httpredir.debian.org jessie-updates Release.gpg
  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]
Err http://security.debian.org jessie/updates Release.gpg
  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]
Err http://apt.armbian.com jessie InRelease
  
Err http://apt.armbian.com jessie Release.gpg
  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]
Reading package lists...
W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/InRelease  

W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie-updates/InRelease  

W: Failed to fetch http://security.debian.org/dists/jessie/updates/InRelease  

W: Failed to fetch http://apt.armbian.com/dists/jessie/InRelease  

W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/Release.gpg  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]

W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie-updates/Release.gpg  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]

W: Failed to fetch http://security.debian.org/dists/jessie/updates/Release.gpg  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]

W: Failed to fetch http://apt.armbian.com/dists/jessie/Release.gpg  Unable to connect to localhost:3142: [IP: 127.0.0.1 3142]
```

I will look into that and try to fix.

zador-blood-stained, 2016-05-28:
> Or some note in documentation that APT_PROXY_ADDR should be provided like 192.168.0.107:3142/. 

`APT_PROXY_ADDR ` was added after user request, it wasn't tested and documented properly. Since now there is separate option for disabling apt-cacher, I'll change the code so you don't need to add  slash at the end of `APT_PROXY_ADDR`

> Above code improve situation, but Updating package list seem to not use proxy

Hm, it may be broken actually. I'll re-check this and push fixes if needed.

pietrushnic, 2016-05-28:
@zador-blood-stained at this point I test [this](https://github.com/3mdeb/lib/commit/bb29c73308ec6405fbf4ed5d9fd349182f435552). Will let you know about results.

pietrushnic, 2016-05-28:
Hmm unfortunately still some problems:

```
[ o.k. ] Applying distribution specific tweaks for [ jessie ]
sed: can't read /root/output/cache/sdcard/etc/ssh/sshd_config: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/default/console-setup: No such file or directory
[ o.k. ] Installing packages [ linux-image-sun8i ]
[ o.k. ] Installing u-boot [ linux-u-boot-orangepione ]
[ o.k. ] Installing headers [ linux-headers-sun8i ]
[ o.k. ] Installing firmware [ linux-firmware-image-sun8i ]
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
cp: cannot create regular file '/root/output/cache/sdcard/etc/lirc/lircd.conf': No such file or directory
[ o.k. ] Installing board support package [ orangepione ]
[ o.k. ] Creating boot scripts [ orangepione ]
[ o.k. ] Installing extra applications and drivers #
[ o.k. ] Building deb [ hostapd ]
[ o.k. ] Downloading sources [ hostapd ]
[ o.k. ] Compiling [ v2.5 ]
[ error ] Not built [ hostapd ]
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:547 ]
[ error ] debootstrap-ng was interrupted #
[ o.k. ] Process terminated #
```

pietrushnic, 2016-05-28:
BTW my command is:

```
./compile.sh BOARD=orangepione PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=default APT_PROXY_ADDR=<host_ip_addr>:3142
```

igorpecovnik, 2016-05-28:
>Hmm unfortunately still some problems:

This indicates deboostrap problems / bad cache file.

zador-blood-stained, 2016-05-28:
@pietrushnic 
Please test [this](https://github.com/zador-blood-stained/lib/commit/a218c142fd109ffa7a5638e005863dfa24188278) - I think single quotes in `apt_extra` parameter were the main problem.

pietrushnic, 2016-05-28:
@zador-blood-stained thanks your fix works. Thanks for excellent support.

zador-blood-stained, 2016-06-02:
Are there any other issues regarding building in Docker (apart from icomplete documentation) or this can be closed?

pietrushnic, 2016-06-02:
@zador-blood-stained this can be close. Thanks for help.

closed 2016-06-02 by pietrushnic