armbian/build #319: [OPi One] Booting problem after building on Debian Sid

pietrushnic, 2016-05-25:
I know that Debian Sid is not supported yet but would like to help enable it, if possible. The problem I'm facing seems to be related to `ext4` filesystem in u-boot. I would like to confirm that other see similar thing. Bootlog looks like below:

```
U-Boot SPL 2016.05-armbian (May 25 2016 - 18:46:20)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2016.05-armbian (May 25 2016 - 18:46:20 +0200) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi One
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...

USB device 0: unknown device
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
=> 
```

It is impossible to list any files in `/boot` directory, but partition seems to be read correctly:

```
=> part list mmc 0  

Partition Map for MMC device 0  --   Partition Type: DOS

Part    Start Sector    Num Sectors     UUID            Type
  1     2048            2914304         a8140803-01     83
=> ext4ls mmc 0 boot
=> ext4ls mmc 0     
<DIR>       4096 .
<DIR>       4096 ..
<DIR>      16384 lost+found
<DIR>       4096 bin
<DIR>          0 boot
<DIR>          0 dev
<DIR>       4096 etc
<DIR>       4096 home
<DIR>          0 lib
<DIR>          0 media
<DIR>       4096 mnt
<DIR>       4096 opt
<DIR>          0 proc
<DIR>          0 root
<DIR>       4096 run
<DIR>       4096 sbin
<DIR>          0 selinux
<DIR>          0 srv
<DIR>       4096 sys
<DIR>       4096 tmp
<DIR>          0 usr
<DIR>          0 var
=> ext4ls mmc 0 tmp
<DIR>       4096 .
<DIR>       4096 ..
```

Some directories (does with 4096) seem to list correctly, but those with 0 just show nothing. I wonder if this is known bug or if there is workaround for this problem.

zador-blood-stained, 2016-05-25:
This is a known issue in new Debian releases (see #279). You probably have to edit `/etc/mke2fs.conf` on your build host to disable 64-bit FS feature by default or change ext4 creation flags like [here](https://github.com/oleid/lib/commit/9cff6956ed4638f8f76410c9e04d0c33142b41c4)

pietrushnic, 2016-05-25:
@zador-blood-stained many thanks for that hint. It works great at least it reads partition correctly, but I still have problems with booting OPi One, will try to debug that.

commit 3305dd45 (2016-05-27): Merge pull request #322 from 3mdeb/sid-fixes

Fixes for building Orange Pi One on Debian Sid (fixes #321, fixes #319)
closed 2016-05-25 by pietrushnic