armbian/build #320: Custom script.bin symbolic link and Updates

Phipli, 2016-05-25:
After running apt-get dist-upgrade the symbolic link to my customised script.bin file was overwritten. In my specific case (possibly due to using an only recently adopted board - the Orange Pi Lite) this caused the machine to lose the ability to set the correct processor speed, and more seriously, freeze following boot.

Considering how autodetection has behaved on similar devices, it is probably worth noting that I was using a USB wifi dongle to perform the dist-upgrade, which might have resulted in a false detection during auto detection, compounding the issue. The issue would have still been present though if correct detection had occurred and I had linked to a custom script.bin file.

zador-blood-stained, 2016-05-25:
Are you sure that symlink was affected? 
Current logic regarding `.bin` files is [here](https://github.com/igorpecovnik/lib/blob/master/makeboarddeb.sh#L49) - renaming `/boot/bin` to `/boot/bin.old` before updating.
 Right now it can't be changed to something like "Declare `/boot/bin/*.bin` files as conffiles and ask to keep or overwrite on upgrade" because this doesn't work well when `/boot` partition if formatted to FAT filesystem.

igorpecovnik, 2016-05-26:
We need to update board configurations - it's a must. But you should find your script under .old ?

Phipli, 2016-05-26:
Following the upgrade, my .bins were in bin.old, and the symlink had been set to orangepih3.bin - which I'm guessing is an autodetect hiccup (because of the USB WiFi). Even if autodetect got the correct platform, it would still have redirected away from a custom script.bin.

igorpecovnik, 2016-05-26:
This auto detect thing is going away. We just haven't found to deal with it.

We can close it as auto detect issue is resolving elsewhere.

closed 2016-05-26 by igorpecovnik