armbian/build #321: [OPi One] continues reboot after error loading journal

pietrushnic, 2016-05-26:
In Debian Sid build of OPi One I have continuous reboot this can be related with applying #319:

```
[   10.323498] JBD2: Unrecognised features on journal
[   10.339599] EXT4-fs (mmcblk0p1): error loading journal
[   20.892446] CPU1: shutdown
[   20.906559] [hotplug]: cpu(2) try to kill cpu(1)
[   20.922822] [hotplug]: cpu1 is killed! .
[   20.944982] CPU2: shutdown
[   20.953496] [hotplug]: cpu(3) try to kill cpu(2)
[   20.964009] [hotplug]: cpu2 is killed! .
[   20.975579] CPU3: shutdown
[   20.983886] [hotplug]: cpu(0) try to kill cpu(3)
[   20.994343] [hotplug]: cpu3 is killed! .
[   21.004875] wdt_enable, write reg 0xf1c20cb8 val 0x000000b0
[   21.101786] Restarting system.
```
It looks like 3.4 kernel doesn't like new way of creating filesystem. Especially it probably doesn't support some features. Is this correct assumption ? Any ideas how to approach that problem ?

zador-blood-stained, 2016-05-26:
You can check filesystem info and features with `tune2fs -l <partition>` and compare output from your image and one of prebuilt images.

pietrushnic, 2016-05-26:
@zador-blood-stained many thanks for getting me the idea what can be wrong. I just submitted PR that should fix all issues. Now I'm able to build bootable image.

commit 3305dd45 (2016-05-27): Merge pull request #322 from 3mdeb/sid-fixes

Fixes for building Orange Pi One on Debian Sid (fixes #321, fixes #319)
closed 2016-05-27 by igorpecovnik