armbian/build #323: Orange pi plus 2 1280x1024 mode

LevdikovDima, 2016-05-27:
I need 1280x1024 on H3 board. I saw similar patch  [https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-default/0026-h3-hdmi-additional-mode-800x480p-from-lobodis-initia.patch](url)

ThomasKaiser, 2016-06-15:
Did you have a look into https://github.com/dni1337/OrangePI-Kernel/commits/master already?

zldo, 2016-07-24:
Try my patch http://pastebin.com/unSKuZee not tested yet :(
modified h3disp http://pastebin.com/Yd5f7FQa
for calculate constants i used http://www.epanorama.net/faq/vga2rgb/calc.html https://www.mythtv.org/wiki/Modeline_Database
for hdmi_bsp_sun8iw7.c this calculator https://yadi.sk/i/N4gSDQY-tatB8

md2000, 2016-08-02:
> Try my patch http://pastebin.com/unSKuZee not tested yet :(
> modified h3disp http://pastebin.com/Yd5f7FQa

The h3disp HDMIMode values do not match the patched values.
They should correspond with the DISP_TV_MOD_* values.

Eg.
                       23|1680x1050) # res
                                #HDMIMode=23 -- Bad!
                                HDMIMode=36
matches
        DISP_TV_MOD_1680_1050P          = 0x24,

I've tried this mode only so far, but it works.
Only h3disp would need to be changed.

zldo, 2016-08-03:
> I've tried this mode only so far, but it works.
I have only HDMI->VGA adapter, and patch did not work - the terrible distortion of the image and its size.

> Only h3disp would need to be changed.
Modified h3disp http://pastebin.com/DfRJXSY4

> Here are most of the new modes: 30 is missing, as are 11 through 22, which I didn't add because I don't have the pll_video values on hand.
i use this post http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=1707

igorpecovnik, 2016-08-03:
HDMI -> VGA adaptors might break the whole thing so try to test this on native HDMI port. I'll try all modes when I get back to office.

md2000, 2016-08-03:
I only have HDMI->DVI at the moment. I tried a couple of the new modes and they work great.

Most of the modes from 1 to 10 have the "overscan problem" for me. I wonder if those would be fixed with  similarly obtained video_timing and ptbl parameters?

zldo, 2016-08-04:
Somebody tried to use this path - https://patchwork.kernel.org/patch/8191881/

igorpecovnik, 2016-08-05:
I checked all modes and those compatible to my monitor (2160p), works. I am closing this issue - if one found a problem, feel free to reopen.

closed 2016-08-05 by igorpecovnik