armbian/build #324: Odroid C1+ fails on clean Jessie server image after sudo apt-get update && upgrade

NanoAkron, 2016-05-28:
H there,

Just installed the Debian Jessie server image on the eMMC, booted fine. Created new user. Copied over SSH keys into ~/.ssh directory. Logged out and logged back in fine.

Then I ran sudo apt-get update, followed by sudo apt-get -y upgrade.

Then sudo dpkg-reconfigure tzdata to set time to London. Then sudo reboot now.

Now the system fails to boot. Two short blue blinks, one longer blue blink, then no more heartbeat LED, just the red LED. Can't log in.

Just odd...any suggestions welcome.

-Nano.

igorpecovnik, 2016-05-29:
I can confirm that 5.12 breaks the system. Temporally solution is make an image, don't upgrade packages and wait for confirmation that it's solved, than make an upgrade.

igorpecovnik, 2016-05-29:
Repository is not updated yet - wait in v5.13 which will be fixed.

closed 2016-05-29 by igorpecovnik

igorpecovnik, 2016-05-31:
Repository updated and new images uploaded.