armbian/build #325: [OPi PC] /var/lib/dpkg/tmp.ci/preinst: [[: not found

kometchtech, 2016-05-28:
I installed what you build in the armbian build tool such as error occurred.
```
-rw-r--r-- 1 kometch kometch   112328 May 29 02:45 linux-firmware-image-sun8i_5.12_armhf.deb
-rw-r--r-- 1 kometch kometch  5801194 May 29 02:45 linux-headers-sun8i_5.12_armhf.deb
-rw-r--r-- 1 kometch kometch 11774084 May 29 02:45 linux-image-sun8i_5.12_armhf.deb
-rw-r--r-- 1 kometch kometch   154810 May 29 02:45 linux-u-boot-orangepipc_5.12_armhf.deb
kometch@ntp01:~/debs$ sudo dpkg -i ./*
[sudo] password for kometch: 
(Reading database ... 48759 files and directories currently installed.)
Preparing to unpack .../linux-firmware-image-sun8i_5.12_armhf.deb ...
Unpacking linux-firmware-image-sun8i (5.12) over (5.11) ...
Preparing to unpack .../linux-headers-sun8i_5.12_armhf.deb ...
Unpacking linux-headers-sun8i (5.12) over (5.11) ...
Preparing to unpack .../linux-image-sun8i_5.12_armhf.deb ...
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
[... 110 lines cut ...]
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
/var/lib/dpkg/tmp.ci/preinst: 22: /var/lib/dpkg/tmp.ci/preinst: [[: not found
Unpacking linux-image-sun8i (5.12) over (5.11) ...
Selecting previously unselected package linux-u-boot-orangepipc-default.
Preparing to unpack .../linux-u-boot-orangepipc_5.12_armhf.deb ...
Unpacking linux-u-boot-orangepipc-default (5.12) ...
Setting up linux-firmware-image-sun8i (5.12) ...
Setting up linux-headers-sun8i (5.12) ...
Compiling headers - please wait ...
Setting up linux-image-sun8i (5.12) ...
update-initramfs: Generating /boot/initrd.img-3.4.112-sun8i
Setting up linux-u-boot-orangepipc-default (5.12) ...
```

igorpecovnik, 2016-05-28:
It should be fixed with this commit:

https://github.com/igorpecovnik/lib/commit/140f01d652bc9d3e80e21dca4963e41ba075ac42

kometchtech, 2016-05-28:
Forget was told are the built environment.

Environment; ubuntu 14.04.4

This is the problem with Ubuntu 16.04?

igorpecovnik, 2016-05-28:
I think it's (was) problem only on Ubuntu 16.04 if you build it, not build environment.

kometchtech, 2016-05-28:
As was mentioned above, this build on Ubuntu 14.04.4 you are.

zador-blood-stained, 2016-05-28:
IMO this should be fixed by adding bash shebangs to pre- and postinstall scripts.
From [here](https://www.debian.org/doc/manuals/developers-reference/best-pkging-practices.html):

> Keep the maintainer scripts as simple as possible. We suggest you use pure POSIX shell scripts. **Remember, if you do need any bash features, the maintainer script must have a bash shebang line.** POSIX shell or Bash are preferred to Perl, since they enable debhelper to easily add bits to the scripts.

igorpecovnik, 2016-05-28:
OK

closed 2016-05-28 by igorpecovnik