armbian/build #326: [Odroid C2] now comes the kernel config options

kometchtech, 2016-05-28:
When trying to build the image of the Odroid C2 and following Kernel compile options asked is it is.

Environment; ubuntu 14.04.4
```
scripts/kconfig/conf --silentoldconfig Kconfig
*
* Restart config...
*
*
* ARM errata workarounds
*
Cortex-A53: 845719: a load might read incorrect data (ARM64_ERRATUM_845719) [N/y/?] (NEW) 
*
* Boot options
*
Default kernel command string (CMDLINE) [console=ttyAMA0] console=ttyAMA0
  Kernel command line type
  > 1. Use bootloader kernel arguments if available (CMDLINE_FROM_BOOTLOADER)
    2. Extend bootloader kernel arguments (CMDLINE_EXTEND)
    3. Always use the default kernel command string (CMDLINE_FORCE)
  choice[1-3]: 1
Build a concatenated Image.gz/dtb by default (BUILD_ARM64_APPENDED_DTB_IMAGE) [N/y/?] n
UEFI runtime support (EFI) [Y/n/?] (NEW) 
*
* Device Tree and Open Firmware support
*
Device Tree runtime unit tests (OF_UNITTEST) [N/y/?] (NEW) 
*
* Device Drivers
*
ARM Versatile Express platform infrastructure (VEXPRESS_CONFIG) [Y/n/?] y
Mali-300/400/450 support (MALI400) [Y/n/m/?] y
  Enable Mali-450 support (MALI450) [N/y/?] (NEW) 
Enable Mali-470 support (MALI470) [N/y/?] (NEW) 
  Enable debug in Mali driver (MALI400_DEBUG) [N/y/?] n
  Enable Mali profiling (MALI400_PROFILING) [N/y/?] n
  Enable UMP support (MALI400_UMP) [N/y/?] n
  Enable Mali dynamically frequency change (MALI_DVFS) [N/y/?] n
  Map dma-buf attachments on attach (MALI_DMA_BUF_MAP_ON_ATTACH) [Y/n/?] y
  Support for shared interrupts (MALI_SHARED_INTERRUPTS) [N/y/?] n
  Power up Mali PMU domains in parallel (MALI_PMU_PARALLEL_POWER_UP) [N/y/?] n
  Using device tree to initialize module (MALI_DT) [N/y/?] n
  Using devfreq to tuning frequency (MALI_DEVFREQ) [N/y/?] (NEW) 
  Make Mali driver very quiet (MALI_QUIET) [N/y/?] n
UMP support (UMP) [N/m/y/?] (NEW) 
Support for ARM architected timer event stream generation (ARM_ARCH_TIMER_EVTSTREAM) [Y/n/?] y
*
* Firmware Drivers
*
Add firmware-provided memory map to sysfs (FIRMWARE_MEMMAP) [N/y/?] (NEW) 
*
* EFI (Extensible Firmware Interface) Support
*
EFI Variable Support via sysfs (EFI_VARS) [N/m/y/?] (NEW) 
*
* File systems
*
Second extended fs support (EXT2_FS) [N/m/y/?] n
Ext3 journalling file system support (EXT3_FS) [N/m/y/?] n
The Extended 4 (ext4) filesystem (EXT4_FS) [Y/n/m/?] y
  Use ext4 for ext2/ext3 file systems (EXT4_USE_FOR_EXT23) [Y/n/?] y
  Ext4 POSIX Access Control Lists (EXT4_FS_POSIX_ACL) [Y/n/?] y
  Ext4 Security Labels (EXT4_FS_SECURITY) [Y/n/?] y
  EXT4 debugging support (EXT4_DEBUG) [Y/n/?] y
JBD2 (ext4) debugging support (JBD2_DEBUG) [N/y/?] n
Reiserfs support (REISERFS_FS) [M/n/y/?] m
  Enable reiserfs debug mode (REISERFS_CHECK) [N/y/?] n
  Stats in /proc/fs/reiserfs (REISERFS_PROC_INFO) [Y/n/?] y
  ReiserFS extended attributes (REISERFS_FS_XATTR) [Y/n/?] y
    ReiserFS POSIX Access Control Lists (REISERFS_FS_POSIX_ACL) [Y/n/?] y
    ReiserFS Security Labels (REISERFS_FS_SECURITY) [Y/n/?] y
JFS filesystem support (JFS_FS) [N/m/y/?] n
XFS filesystem support (XFS_FS) [N/m/y/?] n
GFS2 file system support (GFS2_FS) [N/m/y/?] n
OCFS2 file system support (OCFS2_FS) [N/m/y/?] n
Btrfs filesystem support (BTRFS_FS) [M/n/y/?] m
  Btrfs POSIX Access Control Lists (BTRFS_FS_POSIX_ACL) [Y/n/?] y
  Btrfs with integrity check tool compiled in (DANGEROUS) (BTRFS_FS_CHECK_INTEGRITY) [N/y/?] n
  Btrfs will run sanity tests upon loading (BTRFS_FS_RUN_SANITY_TESTS) [Y/n/?] y
  Btrfs debugging support (BTRFS_DEBUG) [N/y/?] n
  Btrfs assert support (BTRFS_ASSERT) [N/y/?] n
NILFS2 file system support (NILFS2_FS) [N/m/y/?] n
Enable POSIX file locking API (FILE_LOCKING) [Y/n/?] y
Dnotify support (DNOTIFY) [Y/n/?] y
Inotify support for userspace (INOTIFY_USER) [Y/n/?] y
Filesystem wide access notification (FANOTIFY) [N/y/?] n
Quota support (QUOTA) [Y/n/?] y
Report quota messages through netlink interface (QUOTA_NETLINK_INTERFACE) [N/y/?] n
Print quota warnings to console (OBSOLETE) (PRINT_QUOTA_WARNING) [Y/n/?] y
Additional quota sanity checks (QUOTA_DEBUG) [N/y/?] n
Old quota format support (QFMT_V1) [N/m/y/?] n
Quota format vfsv0 and vfsv1 support (QFMT_V2) [Y/n/m/?] y
Kernel automounter version 4 support (also supports v3) (AUTOFS4_FS) [M/n/y/?] m
FUSE (Filesystem in Userspace) support (FUSE_FS) [M/n/y/?] m
  Character device in Userspace support (CUSE) [N/m/?] n
Overlay filesystem support (OVERLAY_FS) [N/m/y/?] (NEW) 
*
* Miscellaneous filesystems
*
Miscellaneous filesystems (MISC_FILESYSTEMS) [Y/n/?] y
  ADFS file system support (ADFS_FS) [N/m/y/?] n
  Amiga FFS file system support (AFFS_FS) [N/m/y/?] n
  eCrypt filesystem layer support (ECRYPT_FS) [M/n/y/?] m
    Enable notifications for userspace key wrap/unwrap (ECRYPT_FS_MESSAGING) [Y/n/?] y
  Apple Macintosh file system support (HFS_FS) [M/n/y/?] m
  Apple Extended HFS file system support (HFSPLUS_FS) [M/n/y/?] m
    HFS+ POSIX Access Control Lists (HFSPLUS_FS_POSIX_ACL) [N/y/?] n
  BeOS file system (BeFS) support (read only) (BEFS_FS) [N/m/y/?] n
  BFS file system support (BFS_FS) [N/m/y/?] n
  EFS file system support (read only) (EFS_FS) [N/m/y/?] n
  Journalling Flash File System v2 (JFFS2) support (JFFS2_FS) [N/m/y/?] n
  LogFS file system (LOGFS) [N/m/y/?] n
  Compressed ROM file system support (cramfs) (OBSOLETE) (CRAMFS) [Y/n/m/?] y
  SquashFS 4.0 - Squashed file system support (SQUASHFS) [M/n/y/?] m
    File decompression options
    > 1. Decompress file data into an intermediate buffer (SQUASHFS_FILE_CACHE)
      2. Decompress files directly into the page cache (SQUASHFS_FILE_DIRECT)
    choice[1-2?]: 1
    Decompressor parallelisation options
    > 1. Single threaded compression (SQUASHFS_DECOMP_SINGLE)
      2. Use multiple decompressors for parallel I/O (SQUASHFS_DECOMP_MULTI)
      3. Use percpu multiple decompressors for parallel I/O (SQUASHFS_DECOMP_MULTI_PERCPU)
    choice[1-3?]: 1
    Squashfs XATTR support (SQUASHFS_XATTR) [Y/n/?] y
    Include support for ZLIB compressed file systems (SQUASHFS_ZLIB) [Y/n/?] y
    Include support for LZO compressed file systems (SQUASHFS_LZO) [Y/n/?] y
    Include support for XZ compressed file systems (SQUASHFS_XZ) [Y/n/?] y
    Use 4K device block size? (SQUASHFS_4K_DEVBLK_SIZE) [N/y/?] n
    Additional option for memory-constrained systems (SQUASHFS_EMBEDDED) [N/y/?] n
  FreeVxFS file system support (VERITAS VxFS(TM) compatible) (VXFS_FS) [N/m/y/?] n
  Minix file system support (MINIX_FS) [N/m/y/?] n
  SonicBlue Optimized MPEG File System support (OMFS_FS) [N/m/y/?] n
  OS/2 HPFS file system support (HPFS_FS) [N/m/y/?] n
  QNX4 file system support (read only) (QNX4FS_FS) [N/m/y/?] n
  QNX6 file system support (read only) (QNX6FS_FS) [N/m/y/?] n
  ROM file system support (ROMFS_FS) [N/m/y/?] n
  Persistent store support (PSTORE) [N/y/?] n
  System V/Xenix/V7/Coherent file system support (SYSV_FS) [N/m/y/?] n
  UFS file system support (read only) (UFS_FS) [N/m/y/?] n
  F2FS filesystem support (EXPERIMENTAL) (F2FS_FS) [M/n/y/?] m
    F2FS Status Information (F2FS_STAT_FS) [Y/n/?] y
    F2FS extended attributes (F2FS_FS_XATTR) [Y/n/?] y
      F2FS Access Control Lists (F2FS_FS_POSIX_ACL) [N/y/?] n
      F2FS Security Labels (F2FS_FS_SECURITY) [N/y/?] n
    F2FS consistency checking feature (F2FS_CHECK_FS) [N/y/?] n
  EFI Variable filesystem (EFIVAR_FS) [N/m/y/?] (NEW) 
```

kometchtech, 2016-05-28:
Stops with an error of the following and press enter in the default.

```
drivers/built-in.o: In function `mali_drm_init':                                                         x  
  x :(.text+0x5a61c): undefined reference to `drm_platform_init'                                             x  
  x :(.text+0x5a644): undefined reference to `drm_platform_init'                                             x  
  x drivers/built-in.o: In function `mali_drm_exit':                                                         x  
  x :(.text+0x5a6c8): undefined reference to `drm_put_dev'                                                   x  
  x drivers/built-in.o: In function `mali_module_exit':                                                      x  
  x :(.text+0x671e0): undefined reference to `mali_platform_device_unregister'                               x  
  x drivers/built-in.o: In function `mali_module_init':                                                      x  
  x :(.text+0x67218): undefined reference to `mali_platform_device_register'                                 x  
  x :(.text+0x6727c): undefined reference to `mali_platform_device_unregister'                               x  
  x drivers/built-in.o:(.data+0x8238): undefined reference to `drm_poll'                                     x  
  x drivers/built-in.o:(.data+0x8240): undefined reference to `drm_ioctl'                                    x  
  x drivers/built-in.o:(.data+0x8250): undefined reference to `drm_mmap'                                     x  
  x drivers/built-in.o:(.data+0x8258): undefined reference to `drm_open'                                     x  
  x drivers/built-in.o:(.data+0x8268): undefined reference to `drm_release'                                  x  
  x make: *** [vmlinux] Error 1                                                                              x  
  mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj  
    

[ error ] ERROR in function compile_kernel [ common.sh:176 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated #
```

igorpecovnik, 2016-05-28:
It works now.
https://github.com/igorpecovnik/lib/commit/96941dd4dd33571c0edc39c704c90d39aaaa99a7

closed 2016-05-28 by igorpecovnik