armbian/build #330: Humming board vanilla kernel boots in to black screen after init

guysoft, 2016-05-30:
Hey,
I have a HummingBoard-Pro (Carrier) with Dual-Core (i2ex)
When I boot using ``Armbian_5.10_Cubox-i_Debian_jessie_4.5.2.raw``  image

This appears on the screen (numbers might not be exact because I had to photograph the screen and type, first line is indeed twice):
```
** File not found /zImage **
** File not found /zImage **
247 bytes read in 100ms (6.8KiB/s)
## Executing script at 1080000
3400648bytes read in 326 ms (12.8 Mib/s)
kernel image @ 0x10800000 [ 0x00000 - x42f728 )
# Flattened Device Tree blob at 1800000
```
Then I get a blank screen. No signal to HDMI.

How I got the image on the sd card:
```
sudo dd if=./Armbian_5.10_Cubox-i_Debian_jessie_4.5.2.raw | pv -s 1296039936 | sudo dd of=/dev/sdc
```
Also confirmed the the image is not corrupt using the gpg sig:
```
$ gpg --verify Armbian_5.10_Cubox-i_Debian_jessie_4.5.2.raw.asc Armbian_5.10_Cubox-i_Debian_jessie_4.5.2.raw
gpg: Good signature from ...
```

igorpecovnik, 2016-05-30:
Yes, this is correct since there is no HDMI driver ... it's written at download page, but yes, it's possible not to see that detail - serial console only. I made image since it's working well - for headless usage.

closed 2016-05-30 by igorpecovnik

guysoft, 2016-05-30:
Oh, ok, thanks.
Perhaps its worth writing in bold "this means you will have no screen"
I think I actually read that and didn't understand the implications.