armbian/build #335: [Cubietruck] ERROR in function compile_kernel

kometchtech, 2016-06-02:
If you try to build with next of Kernel 4.6.1 cubietruck, it will be in the following such errors.

```
   LD [M]  drivers/media/usb/gspca/stv06xx/gspca_stv06xx.o              │
  │   LD      drivers/media/usb/built-in.o                                 │
  │   LD      drivers/media/built-in.o                                     │
  │ make: *** [drivers] Error 2                                            │
  └────────────────────────────────────────────────────────────────────────┘

[ error ] ERROR in function compile_kernel [ common.sh:176 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated #
```
ENV:Ubuntu 14.04.4

igorpecovnik, 2016-06-02:
It's building now.
https://github.com/igorpecovnik/lib/commit/c788fc379d4fabd5453cc490fb03865c67fe71fa

closed 2016-06-02 by igorpecovnik

kometchtech, 2016-06-02:
Sure you can build without any problems.