armbian/build #337: 50unattended-upgrades: "CODENAME-Security" doesn't provide Security patches under Debian Jessie

BernieO, 2016-06-03:
At least for Debian Jessie (I use Armbian-Jessie on Banana Pi) Security-Updates are not delivered via `l=jessie-Security` but via `l=Debian-Security`
I assume the same applies to Debian Wheezy.
So the last line in lib/scripts/50unattended-upgrades needs to be changed.
I don't know though about Trusty or Xenial...

zador-blood-stained, 2016-06-19:
@igorpecovnik 

I unpacked `unattended-upgrades` package for all 4 distributions, and it seems like `50unattended-upgrades` file has correct settings by default for wheezy, jessie and trusty and it is not installed in xenial.

IMO we don't need to provide our own file and we should recommend purging and reinstalling `unattended-upgrades` package after 5.15 release (I'll pack proper file for Xenial in board support packages).

igorpecovnik, 2016-06-19:
OK, so we can ditch https://github.com/igorpecovnik/lib/blob/master/scripts/50unattended-upgrades
and when you add for Xenial, this is fixed?

zador-blood-stained, 2016-06-19:
I checked Xenial build, and it creates config files at postinstall stage, so we don't need to touch `50unattended-upgrades`.
```
Setting up unattended-upgrades (0.90) ...
Creating config file /etc/apt/apt.conf.d/20auto-upgrades with new version
Creating config file /etc/apt/apt.conf.d/50unattended-upgrades with new version
```
I'll check [02periodic](https://github.com/igorpecovnik/lib/blob/master/scripts/02periodic) settings, and if they are OK, this will be closed.

zador-blood-stained, 2016-06-19:
`02periodic` looks OK, log entry was added to "[Prepare v5.15](http://forum.armbian.com/index.php/topic/1425-prepare-v515/)" thread.

closed 2016-06-19 by zador-blood-stained