armbian/build #338: [odroid C2] can't login with SSH

kometchtech, 2016-06-03:
I have created the image of the 3.14.71patch kernel have been applied, in SSH can't login to the Odroid C2 confirmed.

Do not have HDMI, so I do not know is the local State.

kometchtech, 2016-06-03:
log, such as the following in dmesg was being output.
```
[    5.458109] libphy: stmmac-0:00 - Link is Up - 1000/Full
[   49.011578] systemd-journald[205]: Failed to write entry (21 items, 609 bytes), ignoring: Invalid argument
[   49.011765] systemd-journald[205]: Failed to write entry (21 items, 587 bytes), ignoring: Invalid argument
[   49.012196] systemd-journald[205]: Failed to write entry (21 items, 569 bytes), ignoring: Invalid argument
[   52.974565] systemd-journald[205]: Failed to write entry (21 items, 603 bytes), ignoring: Invalid argument
[   52.974682] systemd-journald[205]: Failed to write entry (21 items, 582 bytes), ignoring: Invalid argument
[   53.090422] systemd-journald[205]: Failed to write entry (21 items, 564 bytes), ignoring: Invalid argument
[   53.391681] systemd-journald[205]: Failed to write entry (21 items, 571 bytes), ignoring: Invalid argument
[   53.391868] systemd-journald[205]: Failed to write entry (21 items, 566 bytes), ignoring: Invalid argument
[   53.392755] systemd-journald[205]: Failed to write entry (21 items, 548 bytes), ignoring: Invalid argument
[   60.182710] systemd-journald[205]: Failed to write entry (20 items, 507 bytes), ignoring: Invalid argument
[   62.181017] systemd-journald[205]: Failed to write entry (20 items, 508 bytes), ignoring: Invalid argument
[   62.181855] systemd-journald[205]: Failed to write entry (22 items, 551 bytes), ignoring: Invalid argument
[   62.181905] systemd-journald[205]: Failed to write entry (23 items, 538 bytes), ignoring: Invalid argument
[   62.182167] systemd-journald[205]: Failed to write entry (26 items, 679 bytes), ignoring: Invalid argument
[   62.221797] systemd-journald[205]: Failed to write entry (21 items, 711 bytes), ignoring: Invalid argument
[   62.221847] systemd-journald[205]: Failed to write entry (19 items, 523 bytes), ignoring: Invalid argument
[   64.725953] systemd-journald[205]: Failed to write entry (22 items, 575 bytes), ignoring: Invalid argument
[   64.726167] systemd-journald[205]: Failed to write entry (22 items, 566 bytes), ignoring: Invalid argument
[   64.728016] systemd-journald[205]: Failed to write entry (26 items, 670 bytes), ignoring: Invalid argument
[   66.143545] systemd-journald[205]: Failed to write entry (20 items, 509 bytes), ignoring: Invalid argument
[   66.143762] systemd-journald[205]: Failed to write entry (20 items, 510 bytes), ignoring: Invalid argument
[   66.144493] systemd-journald[205]: Failed to write entry (22 items, 551 bytes), ignoring: Invalid argument
```

zoic21, 2016-06-04:
Hi,
I have the same issue, did you find a solution ?

zador-blood-stained, 2016-06-04:
Found this issue: [link](http://forum.odroid.com/viewtopic.php?t=21580&p=144061#p144450)

You'll have to update your `/boot/boot.ini` file with [these](https://github.com/igorpecovnik/lib/commit/e8d388a98b09ff31a61c260a37da301754751a57) changes (use card reader on another machine or boot from another OS/use microSD adapter to update eMMC).

@igorpecovnik 
You may want to release new image for C2 ASAP. Once new image is uploaded this issue can be closed.

kometchtech, 2016-06-05:
Only boot.ini of rewriting the situation did not change.

It was to re-create the image, the problem did not occur.

igorpecovnik, 2016-06-05:
https://github.com/igorpecovnik/lib/commit/d45ad4af793b2c21920ea8299bf0364d9eab7fea

closed 2016-06-05 by igorpecovnik