armbian/build #339: PMU Level Inconsistant in some fex

seblucas, 2016-06-04:
Hi,

I saw that Zador fixed some H3 fex files to avoid using octal value for pmu_level1 (see https://github.com/igorpecovnik/lib/commit/d6185b560000dda25029727464b8a262d23ee0f9). I saw that some H3 fex file were not modified : https://github.com/igorpecovnik/lib/blob/master/config/fex/orangepiplus.fex#L1115 for example. 

Is it normal ?

The following search show 3 fex files ; https://github.com/igorpecovnik/lib/search?utf8=%E2%9C%93&q=01100

seblucas, 2016-06-04:
Forgot to say that I can make a PR if you're ok with the change

zador-blood-stained, 2016-06-04:
These values (`pmu_level0` and `pmu_level1`) are used only with GPIO regulator (`pmuic_type = 1`).

seblucas, 2016-06-04:
OK, sorry for the noise then ;)

closed 2016-06-04 by seblucas