armbian/build #341: [RfE] nand-sata-install.sh: Install on NAND/eMMC but boot from SD card?

ThomasKaiser, 2016-06-05:
Since we're seeing some new sunxi boards shipping with onboard eMMC while prices for good large SD cards continually are decreasing I wonder whether we should implement an alternative `nand-sata-install.sh` installation mode: Being able to run the rootfs off from eMMC (or NAND) while using a large SD card for additional storage.

Since on all sunxi boards we tested so far SD card has higher boot priority it would be desirable if the installer would modify `boot.cmd` on SD card in a way that the rootfs can reside on eMMC and that /boot/.verbose will also be read from there.

But since I've not the slightest idea how all the other platforms behave we also support I thought I ask first. Also a first test trying to let u-boot read from `ext4load mmc 1` instead of `ext4load mmc 0` failed (but to be honest: that was just trial&error and I'm an absolute u-boot noob!)

So this whole issue is created from a very 'sunxi only' perspective and asks for enlightenment what we have to consider if we would try to provide this installation variant (which will become more and more desirable I would believe).

ThomasKaiser, 2016-06-10:
An alternative would be in addition to `nand-sata-install.sh` working the usual way another script that can be used to prepare 1-n SD cards _after_ the installation has been moved to eMMC/NAND. This script would copy the necessary bootloader parts from eMMC/NAND to the SD card in question, creates there two partitions, copies the `/boot/` contents over and modifies boot.cmd there so the user ends up with an installation that boots from eMMC/NAND and also from this insertes SD card in the same way.

Drawback: contents of `/boot/` on eMMC/NAND and on the `/boot/` partition on the SD card might get out of sync. But the alternative (only putting u-boot+spl on the SD card and let u-boot read everything it needs from `/boot/` from eMMC/NAND -- kernel, `script.bin`/`.dtb` and `.verbose` -- from there) at least didn't work for me. But maybe `ext4load mmc 1` is wrong syntax anyway or u-boot needs patches to be able to access not only the first mmc device?

ThomasKaiser, 2016-06-10:
BTW: It would also be a good idea to repeat the `rsync` without user notification call directly after the [current invocation](https://github.com/igorpecovnik/lib/blob/master/scripts/nand-sata-install/usr/lib/nand-sata-install/nand-sata-install.sh#L56-L57) since this will run pretty fast and also minimizes changes that might happened at the source since the initial `rsync` run might take some time and source and destination got out of sync in the meantime.

At least that's what I do always when migrating large server volumes. Letting rsync run periodically before the final switch migrating TB of data to be able to let the final rsyncs run in pretty short time after the fileserver daemons have been stopped at source and destination.

No PR with this enhancement since I wait for any feedback on `nand-sata-install.sh` first.

zador-blood-stained, 2016-06-10:
Ideally I would propose using "root by PARTUUID" in nand-sata-install script and, ideally, in new images by default. This should also solve SD/eMMC numbering issues in u-boot. See #117 for some info and [this](http://forum.armbian.com/index.php/topic/853-armbian-customization/page-3#entry7939) thread.

> But since I've not the slightest idea how all the other platforms behave we also support I thought I ask first. Also a first test trying to let u-boot read from ext4load mmc 1 instead of ext4load mmc 0 failed (but to be honest: that was just trial&error and I'm an absolute u-boot noob!)

Boot device is always "mmc 0". So kernel, initrd, boot script and DT/.bin files should be loaded from boot device, after that kernel will search rootfs by PARTUUID on all supported devices (SD/eMMC/USB/SATA/NAND)

I'll make some tests later with H3 boards, but I believe that you can use SD as storage if it doesn't have SPL signature - this way even if card is inserted at boot, BROM doesn't find SPL on it and proceeds to next device (eMMC)

zador-blood-stained, 2016-06-21:
This scenario also will need an enhancement to boot script and u-boot postinstall script since currently u-boot is always installed to `/dev/mmcblk0` and u-boot package upgrade may write u-boot on SD instead of eMMC thus bricking this setup until you pull SD card out.

ThomasKaiser, 2016-06-21:
Hmm... it gets rather complicated. Could be walking through every available `/dev/mmcblk*` device be an option checking for SPL signatures there and installing/updating it where one is already present?

zador-blood-stained, 2016-06-21:
Main complication for me is we **have to** make boot scripts updateable without breaking anything. So we have to separate configuration bits (verbosity, root device, extra kernel command line parameters) from actual logic. After that features like root on whatever device, switching CLI/desktop and back will be easier to implement.

As for finding where to write u-boot - boot device is always recognized as "mmc 0", so we need to get PARTUUID for mmc 0:1 and append it to kernel command line in boot script, then in postinstall script we can read kernel command line and find device by PARTUUID.

jernejsk, 2016-06-21:
If I understand correctly, the problem is how to know from which interface board booted? That can be easily solved on sunxi board by reading special memory location. Check here:
http://git.denx.de/?p=u-boot/u-boot-sunxi.git;a=blob;f=arch/arm/include/asm/arch-sunxi/spl.h;h=a966a888a7f97dedc986f56bb418befcfd83170e;hb=refs/heads/next#l60 (line 60) and
http://comments.gmane.org/gmane.comp.hardware.netbook.arm.sunxi/23553

Script could be written based on those informations. For example, if board is booted from mmc0 (sd card) and U-Boot could try to load some file from eMMC. If everything is fine, it would change bootcmd to match situation accordingly. That way you have universal U-Boot script which would be same for SD cards and eMMC. Also it wouldn't be locked to special UUID, although this might be desirable...

zador-blood-stained, 2016-06-21:
@jernejsk 
U-boot / SPL uses this and it assigns internal number 0 to MMC interface it booted from. We are talking about:

- Extending NAND/SATA/USB/eMMC install script and default boot script to use UUID/PARTUUID for rootfs

- When we are updating u-boot via apt-get, currently postinstall script always writes it to /dev/mmcblk0, this needs to be changed to handle different scenarios.

- We can't just replace boot script to a newer version because it will break brick setups with rootfs on anything other than SD card

> Also it wouldn't be locked to special UUID, although this might be desirable...

U-boot can read partuuid and store it into variable, then you can pass it to kernel command line, so you don't need to hardcode anything into boot script.

neomanic, 2016-07-05:
I've been doing some testing with my new Lime2-eMMCs and SD cards, as we wish to set up our systems to boot from EMMC and load storage from SATA, but then have the SD card slot free for emergency recovery if required.

So, I can confirm zador's guess that if the SD card doesn't have an SPL signature, then it is ignored in the boot process... the system boots fine from EMMC even with an SD card in the slot. 

(Note that for kernel params I am using root=PARTUUID=xxx)

zador-blood-stained, 2016-10-22:
Since now we should (hopefully) have support for updating u-boot on eMMC without touching SD card, this can be resolved by optinally erasing SPL signature from SD at the end of eMMC installation in nand-sata-install.
Something like
```
dd if=/dev/zero of=/dev/mmcblk0 bs=1024 seek=8 count=1
```
where /dev/mmcblk0 is the device current OS was booted from

zador-blood-stained, 2016-10-22:
Also there is a question for u-boot update procedure: now updating u-boot on eMMC is possible, but if we enable this by default, then we may end up erasing u-boot on eMMC even if we don't want to do this (i.e. Armbian running from SD and Android installed on eMMC)

ThomasKaiser, 2016-10-23:
I would believe that https://github.com/igorpecovnik/lib/commit/44f20bfcb0211d939749338db9bbdd7c3def9ce5 answered u-boot question already by yourself? What about adding `| sort` to `lsblk` call to prevent different behaviour in case `lsblk` authors change output order for whatever reasons in the future?

    root@beelinkx2:/home/tk# lsblk -d -n -p -o NAME
    /dev/mmcblk0boot0
    /dev/mmcblk0boot1
    /dev/mmcblk0
    /dev/mmcblk1
    root@beelinkx2:/home/tk# lsblk -d -n -p -o NAME | sort
    /dev/mmcblk0
    /dev/mmcblk0boot0
    /dev/mmcblk0boot1
    /dev/mmcblk1

Maybe outputting the `dd if=/dev/zero of=/dev/mmcblk0 bs=1024 seek=8 count=1` recommendation from within `nand_sata_install` would be an idea? Also suggesting to [change `mmc0`'s `sdc_detmode` from 3 to 1 in fex files](http://forum.armbian.com/index.php/topic/1702-orange-pi-plus-2e-where-is-16ghz-and-sd/?p=13163) on legacy images might be worth an idea?

zador-blood-stained, 2016-10-23:
> I would believe that 44f20bf answered u-boot question already by yourself?

I just disabled the dangerous part by default, replacing it with a more reliable solution (but only for new images with new boot script). Sorting is not needed here since we don't rely on device order but on boot0 "magic" signature. For eMMC devices with boot partitions additional code may be required, but this will be tested in the future.

> Maybe outputting the dd if=/dev/zero of=/dev/mmcblk0 bs=1024 seek=8 count=1 recommendation from within nand_sata_install would be an idea?

If you don't want implementing "Install on NAND/eMMC but boot from SD card" as a dialog option before or after installation, then we need to at least document SPL erasing procedure.

> Also suggesting to change mmc0's sdc_detmode from 3 to 1 in fex files on legacy images might be worth an idea?

Yes, this needs to be done.

ThomasKaiser, 2016-10-23:
Tried to address both with https://github.com/igorpecovnik/lib.docs/commit/693a9582cad385e63562d7617e3567883581cdc0

ThomasKaiser, 2016-10-23:
Closing this since current situation should be sufficient.

closed 2016-10-23 by ThomasKaiser