armbian/build #345: linux-headers-odroidc2 make scripts fail

gklipel, 2016-06-05:
Hi all!!!

I'm triyng to install xtables-addons-dkms and it fails claiming that some headers are not present, and in fact, it isn't.

`cat /var/lib/dkms/xtables-addons/2.6/build/make.log
DKMS make.log for xtables-addons-2.6 for kernel 3.14.70-odroidc2 (aarch64)
Sun Jun  5 23:45:11 CEST 2016
make: Entering directory '/usr/src/linux-headers-3.14.70-odroidc2'
Makefile:590: /usr/src/linux-headers-3.14.70-odroidc2/arch//Makefile: No such file or directory
make: *** No rule to make target '/usr/src/linux-headers-3.14.70-odroidc2/arch//Makefile'.  Stop.
make: Leaving directory '/usr/src/linux-headers-3.14.70-odroidc2'`

So I updated the xtables build-aux/config.guess to guess what arch is it going to compile to, and it works, but, it claims some headers that wasn't present. One example was the directory "arch/arm/include", it only exists on arm64 arch.

Well, so when I ran make scripts, shows this up:

`root@odroidc2:/lib/modules/3.14.70-odroidc2/build# make scripts
  CC      scripts/mod/empty.o
  MKELF   scripts/mod/elfconfig.h
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTLD  scripts/mod/modpost
  HOSTCC  scripts/selinux/genheaders/genheaders
scripts/selinux/genheaders/genheaders.c:13:22: fatal error: classmap.h: No such file or directory
 #include "classmap.h"
                      ^
compilation terminated.
scripts/Makefile.host:118: recipe for target 'scripts/selinux/genheaders/genheaders' failed
make[3]: *** [scripts/selinux/genheaders/genheaders] Error 1
scripts/Makefile.build:455: recipe for target 'scripts/selinux/genheaders' failed
make[2]: *** [scripts/selinux/genheaders] Error 2
scripts/Makefile.build:455: recipe for target 'scripts/selinux' failed
make[1]: *** [scripts/selinux] Error 2
Makefile:529: recipe for target 'scripts' failed
make: *** [scripts] Error 2
`

Any thoughts on this?? I'm not getting any luck on this...

Thanks!

gklipel, 2016-06-05:
In time, another thing very useful would be add headless support on boot.ini. I've noticed that on latest http://oph.mdrjr.net/ kernel images it is possible. It would be too much to ask this improve?

gklipel, 2016-06-06:
Well, it turns out that on linux-headers-odroidc2 package we don't have all the include files required to make scripts properly. These files are on:
scripts/selinux/include
Strangely these files (*.h) had to been copied to scripts/selinux/genheaders and scripts/selinux/mdp
After, other files gotten from hardkernel's 3.14.y-odroidc2 were copied to tools/include/tools, specifically be_byteshift.h and le_byteshift.h, then running make oldconfig && make prepare && make scripts went smooth... But, yet I'd faced another problem on compiling xtables-common-dkms... The modules generated won't load. The error can be overriden by appending on section "EXTRAVERSION = " the expression "-odroidc2" and then everything worked like a charm.

These steps were taken to add xtables geoip and ipp2p modules running, so, if anyone want to add these resources on iptables, just do this steps and you'll get it working.

Don't know if these files could be added to header package as a update, but it would be great to count on it.

Thanks!

zador-blood-stained, 2016-06-07:
> In time, another thing very useful would be add headless support on boot.ini. I've noticed that on latest http://oph.mdrjr.net/ kernel images it is possible. It would be too much to ask this improve?

Added: c51bd13341fc0c069191f5e624089699b18a6186
Right now boot script is not upgraded automatically, but this will be changed soon (hopefully)

> Well, it turns out that on linux-headers-odroidc2 package we don't have all the include files required to make scripts properly.

Should be fixed: 98fca07d8906555ebd419e2cd3eb2a5c0af80298

closed 2016-06-30 by zador-blood-stained