armbian/build #347: /boot/initrd.img-3.4.112-sun8i does not exist. Cannot update.

avinashga23, 2016-06-07:
Hi,

When i tried to upgrade using apt-get i am getting /boot/initrd.img-3.4.112-sun8i does not exist. Cannot update. at the end. How to get rid of this message? Is this an issue with upgrade?

Regards,
--
Avinash

igorpecovnik, 2016-06-07:
Can you point me to the image (version?) from which you want to upgrade?

avinashga23, 2016-06-07:
I was in 5.11 and recently upgraded to 5.13 via apt-get for my both orange pi pc and orange pi plus 2. Today I did apt-get upgrade to check for any updates and it went well but in the end I got this message.

seblucas, 2016-06-07:
I had the same error and posted in the forum : http://forum.armbian.com/index.php/topic/1015-nanopi-m1/#entry10059

ThomasKaiser, 2016-06-07:
So how does the output of `sudo armbianmonitor -u` looks like?

BTW: @seblucas the reason some H3 variants are already at 5.14 is related to onboard eMMC (see [here](https://github.com/igorpecovnik/lib/commit/3907a66cf39a4289d58310bb8d400b556d9271e6) and [here](https://github.com/igorpecovnik/lib/commit/2a2e1904b236bfcfd61f7e41a32a200bfa676b50))

seblucas, 2016-06-07:
See here : http://sprunge.us/ehCR

ThomasKaiser, 2016-06-07:
Yes, at least that's wrong but kernel update suceeded. Can you please check where /boot/script.bin links to? Thanks for the log anyway, we will need to update/upload a bit more information.

Regarding the initrd stuff I've no idea (whether it's safe to ignore since it might only work correctly on OS images build after [this commit](https://github.com/igorpecovnik/lib/commit/6554d1addf3f0db5068fa3db212ac33ad85c8f0e) and throws error messages on earlier ones)

ThomasKaiser, 2016-06-07:
@seblucas Since you seem to own a NanoPi M1 could you look through the [DRAM reliability thread](http://forum.armbian.com/index.php/topic/1322-testers-wanted-testing-dram-reliability-on-bpi-m2-and-nanopi-m1/) and provide results in the very same thread? Also please the exact type of DRAM would be great.

seblucas, 2016-06-07:
here it is : 

```
root@nanopim1:/boot# ls -la /boot
total 10220
drwxr-xr-x  4 root root    4096 Jun  6 17:05 .
drwxr-xr-x 22 root root    4096 Jun  6 08:48 ..
drwxr-xr-x  2 root root    4096 Jun  6 09:06 bin
drwxr-xr-x  2 root root    4096 Jun  6 08:48 bin.old
-rw-r--r--  1 root root    6944 Apr 30 20:39 boot.bmp
-rw-r--r--  1 root root    2372 Apr 30 20:39 boot.cmd
-rw-r--r--  1 root root    2444 Apr 30 20:40 boot.scr
-rw-r--r--  1 root root   95050 May 31 19:00 config-3.4.112-sun8i
lrwxrwxrwx  1 root root      22 Jun  6 08:59 script.bin -> /boot/bin/nanopim1.bin
-rw-r--r--  1 root root 2128821 May 31 19:00 System.map-3.4.112-sun8i
-rw-r--r--  1 root root 3132344 Jun  6 09:07 uInitrd
-rw-r--r--  1 root root     189 Jun  6 17:05 .verbose
-rwxr-xr-x  1 root root 5064616 May 31 19:00 vmlinuz-3.4.112-sun8i
lrwxrwxrwx  1 root root      21 Jun  6 09:07 zImage -> vmlinuz-3.4.112-sun8i
```

ThomasKaiser, 2016-06-07:
> I manually fixed the symlink before upgrading

Hmm... it starts to get confusing. According to your [forum post](http://forum.armbian.com/index.php/topic/1015-nanopi-m1/?view=getlastpost) the correct packages (5.11 for NanoPi M1 that should fix the ARISC errors) were loaded. So what did you fixed manually and why? Would be great if you provide the output of

    curl -F 'sprunge=<-' http://sprunge.us </var/log/apt/history.log

(in case log rotation already happened maybe the last log also -- using zcat and curl)

@igorpecovnik Any chance to bring NanoPi M1 to 5.14 too? And the other H3 board where download section still provides a 5.10 image?

seblucas, 2016-06-07:
The only Nanopi release on armbian.com is 5.10 (I posted [here](http://forum.armbian.com/index.php/topic/1355-inconsistancies-in-armbian-version-across-h3-devices/) about that ). So I changed the hostname and fixed the symlink (and also downloaded the latest nanopi fex) before trying to upgrade.

seblucas, 2016-06-07:
Sorry forgot the sprunge : http://sprunge.us/IVgM

It seems complete

ThomasKaiser, 2016-06-07:
Ah, now I understand. The old OS images relied on our broken auto detection feature. According to `/var/log/apt/history.log` the following Armbian packages have been updated

    linux-jessie-root-nanopim1:armhf (5.10, 5.11)
    linux-u-boot-nanopim1-default:armhf (5.10, 5.11)

Can't say that much about this since I'm not familiar with the online upgrade stuff and can't test also at the moment. I hope the others jump in.

@igorpecovnik Do you already have a plan to store the current Armbian version somewhere? Since I would believe the information in `/etc/armbian.txt` reflects _build time_ and not current version?

seblucas, 2016-06-07:
If I may, instead of storing you can get it from there : 

```
root@nanopim1:/boot# dpkg -l | grep linux-image
ii  linux-image-sun8i              5.13                      armhf        Linux kernel, version 3.4.112-sun8i
```

That's what I do on my banana pi.

igorpecovnik, 2016-06-07:
Nanopi images were rebuilt and will be available in download section in matter of minutes. Regarding version, best is to store armbian.txt to the update package and we are done. I am working on it.

Updating procedures. They should work and to find a possible bug I would need to recreate and catch it. If this is referred to upgrade from first testing image of H3 boards is useless .. they are prerelease and contain more bugs and upgrade might not cover all.

igorpecovnik, 2016-06-07:
One idea - some package names could be different between preview and current versions. This could cause such problems.

avinashga23, 2016-06-07:
I did 
sudo update-initramfs -c -k 3.4.112-sun8i
and 
sudo update-initramfs -u -k 3.4.112-sun8i

and tried again with apt-get update and upgrade. this time i did not get this message. It seems the file get generated by above commands.

ThomasKaiser, 2016-06-07:
> One idea - some package names could be different between preview and current versions. This could cause such problems.

This was also my idea. But then we have to add that to documentation that H3 images prior to $what might not be able to reliably upgrade.

Updating `/etc/armbian.txt` from within the apr repository seems also like a good idea to me. So we can use this version from within `armhwinfo` to be stored in the log and can also use `dpkg -l` to check for potential inconsistencies and log them to `armhwinfo.log` too. So in case users are reporting upgrade troubles we might get more info from the log. Also adding the output from `dpkg -l` to armbianmonitor upload seems like a good idea to me.

zador-blood-stained, 2016-06-07:
> Do you already have a plan to store the current Armbian version somewhere? Since I would believe the information in /etc/armbian.txt reflects build time and not current version?

7f45d2e2aee7d85eb43eed0e9c34836c61d27626

Regarding initramfs: first - it's not a good idea to remove initrd.img after wrapping it in uInitrd **or** remove it on kernel upgrade; second - 6554d1addf3f0db5068fa3db212ac33ad85c8f0e adds proper way of wrapping initrd.img to uInitrd, so this action should be removed from all packaging patches.

igorpecovnik, 2016-06-07:
> 6554d1a adds proper way of wrapping initrd.img to uInitrd, so this action should be removed from all packaging patches.

Should this work everywhere? Wheezy, Jessie, ...?

zador-blood-stained, 2016-06-07:
Yes, I checked `initramfs-tools` code on Wheezy, Trusty and Jessie. Should work in other releases too.

igorpecovnik, 2016-06-07:
OK, than I'll update patches.

seblucas, 2016-06-08:
@avinashga23 
I think this could be closed, what do you think ?

closed 2016-06-08 by igorpecovnik