armbian/build #350: Build all is quite unhappy

faddat, 2016-06-07:
Building all leads to a crash during the first board (armada) 

This is all I have for now, and will post more as I dig into this some.

igorpecovnik, 2016-06-08:
Working fine for me, probably 4.6.1 was broken, we got 4.6.2 today and I found no problems. Closing.

igorpecovnik, 2016-06-08:
BTW: please provide a log where it breaks. It could be wrong compiler issues. This part is not fully fixed.

faddat, 2016-06-08:
oh sure!  It is a compiler issue in fact.  I have a compiler > 5.0 and it tells me I do not.  Updating my code on this side, and I'll post a log.

faddat, 2016-06-08:
```
Your branch is up-to-date with 'origin/master'.
#   BOARD                BRANCH     RELEASE    DESKTOP   
1   armada               default    jessie     no        
2   armada               next       jessie     no        
3   armada               default    trusty     no        
4   armada               next       trusty     no        
5   bananapi             default    jessie     no        
6   bananapi             next       jessie     no        
7   bananapi             default    trusty     no        
8   bananapi             next       trusty     no        
9   bananapi             default    jessie     yes       
10  bananapi             next       jessie     yes       
11  bananapim2           next       jessie     no        
12  bananapim2           next       trusty     no        
13  bananapim2           next       jessie     yes       
14  bananapim2plus       default    jessie     no        
15  bananapim2plus       default    jessie     yes       
16  bananapipro          default    jessie     no        
17  bananapipro          next       jessie     no        
18  bananapipro          default    trusty     no        
19  bananapipro          next       trusty     no        
20  bananapipro          default    jessie     yes       
21  bananapipro          next       jessie     yes       
22  beelinkx2            default    jessie     no        
23  beelinkx2            default    jessie     yes       
24  cubieboard2          default    jessie     no        
25  cubieboard2          next       jessie     no        
26  cubieboard2          default    trusty     no        
27  cubieboard2          next       trusty     no        
28  cubieboard2          default    jessie     yes       
29  cubieboard2          next       jessie     yes       
30  cubieboard           default    jessie     no        
31  cubieboard           next       jessie     no        
32  cubieboard           default    trusty     no        
33  cubieboard           next       trusty     no        
34  cubieboard           default    jessie     yes       
35  cubieboard           next       jessie     yes       
36  cubietruck           default    jessie     no        
37  cubietruck           next       jessie     no        
38  cubietruck           default    trusty     no        
39  cubietruck           next       trusty     no        
40  cubietruck           default    jessie     yes       
41  cubietruck           next       jessie     yes       
42  cubox-i              default    jessie     no        
43  cubox-i              next       jessie     no        
44  cubox-i              default    trusty     no        
45  cubox-i              next       trusty     no        
46  cubox-i              default    jessie     yes       
47  guitar               default    jessie     no        
48  guitar               default    jessie     yes       
49  lamobo-r1            default    jessie     no        
50  lamobo-r1            next       jessie     no        
51  lamobo-r1            default    trusty     no        
52  lamobo-r1            next       trusty     no        
53  lime2                default    jessie     no        
54  lime2                next       jessie     no        
55  lime2                default    trusty     no        
56  lime2                next       trusty     no        
57  lime2                default    jessie     yes       
58  lime2                next       jessie     yes       
59  lime-a10             default    jessie     no        
60  lime-a10             next       jessie     no        
61  lime-a10             default    trusty     no        
62  lime-a10             next       trusty     no        
63  lime-a10             default    jessie     yes       
64  lime-a10             next       jessie     yes       
65  lime                 default    jessie     no        
66  lime                 next       jessie     no        
67  lime                 default    trusty     no        
[... 60 lines cut ...]
128 udoo                 next       trusty     no        
129 udoo                 default    jessie     yes       
130 udoo                 next       jessie     yes       
131 udoo-neo             default    jessie     no        
132 udoo-neo             default    trusty     no        

132 total

[ o.k. ] Building [ Board: armada Kernel:default Release: jessie Desktop: no ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Starting Armbian build script [ @host ]
[ o.k. ] Building [ Armbian 5.14 Armada Debian jessie default ]
[ o.k. ] Syncing clock [ host ]
[ o.k. ] source downloading [ @host ]
[ o.k. ] ... you have latest sources [ u-boot-armada u-boot-2013.01-15t1-clearfog ]
[ o.k. ] ... you have latest sources [ linux-armada linux-3.10.70-15t1-clearfog ]
[ o.k. ] ... you have latest sources [ sunxi-tools  ]
[ o.k. ] ... you have latest sources [ sunxi-display-changer  ]
[ o.k. ] Cleaning [ u-boot-armada/u-boot-2013.01-15t1-clearfog ]
[ o.k. ] Cleaning [ linux-armada/linux-3.10.70-15t1-clearfog ]
[ o.k. ] Cleaning /root/git/output/debs for [ armada default ]
[ error ] ERROR in function find_toolchain [ common.sh:240 ]
[ error ] Could not find required toolchain [ arm-linux-gnueabi-gcc < 5.0 ]
[ o.k. ] Process terminated 
```

igorpecovnik, 2016-06-08:
OK, will reopen since this needs to be fixed.

zador-blood-stained, 2016-06-08:
@faddat 
Do you have the same issue if you select armada default in normal scenario?

ThomasKaiser, 2016-06-08:
BTW: I tried to build Xenial for ODROID-C1:

     [ o.k. ] Using user configuration override [ userpatches/lib.config ]
     [ o.k. ] Starting Armbian build script [ @host ]
     [ o.k. ] Building [ Armbian 5.14 Odroidc1 Ubuntu xenial default ]
     [ o.k. ] Syncing clock [ host ]
     [ o.k. ] source downloading [ @host ]
     [ o.k. ] ... creating a shallow clone [ u-boot-odroidc1 odroidc-v2011.03 ]
     Cloning into '/var/git/Armbian/sources/u-boot-odroidc1/odroidc-v2011.03'...
     remote: Counting objects: 7949, done.
     remote: Compressing objects: 100% (5834/5834), done.
     remote: Total 7949 (delta 2020), reused 7887 (delta 2004), pack-reused 0
     Receiving objects: 100% (7949/7949), 30.64 MiB | 5.05 MiB/s, done.
     Resolving deltas: 100% (2020/2020), done.
     Checking connectivity... done.
     [ o.k. ] ... creating a shallow clone [ linux-odroidc1 odroidc-3.10.y ]
     Cloning into '/var/git/Armbian/sources/linux-odroidc1/odroidc-3.10.y'...
     remote: Counting objects: 52523, done.
     remote: Compressing objects: 100% (48013/48013), done.
     remote: Total 52523 (delta 5731), reused 28608 (delta 3870), pack-reused 0
     Receiving objects: 100% (52523/52523), 139.16 MiB | 9.88 MiB/s, done.
     Resolving deltas: 100% (5731/5731), done.
     Checking connectivity... done.
     [ o.k. ] ... you have latest sources [ sunxi-tools  ]
     [ o.k. ] ... you have latest sources [ sunxi-display-changer  ]
     [ o.k. ] Cleaning [ u-boot-odroidc1/odroidc-v2011.03 ]
     [ o.k. ] Cleaning [ linux-odroidc1/odroidc-3.10.y ]
     [ o.k. ] Cleaning /var/git/Armbian/output/debs for [ odroidc1 default ]
     [ error ] ERROR in function find_toolchain [ common.sh:240 ]
     [ error ] Could not find required toolchain [ arm-linux-gnueabihf-gcc < 4.9 ]
     [ o.k. ] Process terminated 

     root@armbian:/var/git/Armbian# grep -v '#' userpatches/lib.config 
     USE_CCACHE=yes

Next try with jessie:

     ./compile.sh PROGRESS_DISPLAY=plain RELEASE=jessie BUILD_DESKTOP=yes EXTENDED_DEBOOTSTRAP=yes ROOTFS_TYPE=ext4 BRANCH=default COMPRESS_OUTPUTIMAGE=no PROGRESS_LOG_TO_FILE=yes

Exactly same error message and stop.

ThomasKaiser, 2016-06-08:
Addendum: Same settings and sun8i build and everything works as expected.

zador-blood-stained, 2016-06-08:
@ThomasKaiser 

> [ error ] Could not find required toolchain [ arm-linux-gnueabihf-gcc < 4.9 ]

Did you unpack linaro armhf toolchain v4.8 to toolchains directory? Did you [activate](http://forum.armbian.com/index.php/topic/881-prepare-v51-v201604/#entry8109) x86 support for this toolchain?

igorpecovnik, 2016-06-08:
I am working to provide auto tool download ... will be ready asap.

ThomasKaiser, 2016-06-08:
@zador-blood-stained 
> Did you unpack linaro armhf toolchain v4.8 to toolchains directory? Did you activate x86 support for this toolchain?

Nope, I played the average user not reading (yet unfinished) documentation. :)

Is adding `i386` architecture something our build system can do automagically without causing any harm?

> I am working to provide auto tool download ... will be ready asap.

I remember having had a hard time trying to get a specific older Linaro toolchain some time ago. Does anyone know whether the license allows mirroring them? At least we should push all the currently needed toolchains into http://archive.org since this keeps them without running into legal issues.

igorpecovnik, 2016-06-08:
https://github.com/igorpecovnik/lib/commit/30c9c89cc38c0a435b225f2d78af5567dc3189d8

I made a copy of current pack to my mirror. Just in case. Closing.

closed 2016-06-08 by igorpecovnik