armbian/build #353: resize2fs failing on ODROID-C1

ThomasKaiser, 2016-06-08:
On ODROID-C1 partitions were enlarged from within `firstrun` but fs will not be expanded. I used the official jessie build that is a few weeks old. When trying out `resize2fs` an error is thrown:

    root@odroidc1:/home/tk# resize2fs /dev/mmcblk0p2
    resize2fs 1.42.12 (29-Aug-2014)
    Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
    old_desc_blocks = 1, new_desc_blocks = 1
    resize2fs: No space left on device While checking for on-line resizing support

And the partitioning looks like this

    root@odroidc1:/home/tk# cat /proc/partitions 
    major minor  #blocks  name
    
     179        0   15159296 mmcblk0
     179        1      32768 mmcblk0p1
     179        2   15125504 mmcblk0p2

While roots like this:

    /dev/mmcblk0p2  1.3G  1.1G  210M  84% /

Unfortunately `/etc/init.d/resize2fs` fails also (I let the output redirect and it look's like [this](http://pastebin.com/XdcFE665) and since `$rootfstype` will be empty nothing will happen). I tried to reproduce it manually and the whole thing fails at the `blkid` call:

    root@odroidc1:/home/tk# lsb_release -cs
    jessie
    root@odroidc1:/home/tk# set -e
    root@odroidc1:/home/tk# device="/dev/"$(lsblk -idn -o NAME | grep mmcblk0)
    root@odroidc1:/home/tk# echo $device
    /dev/mmcblk0
    root@odroidc1:/home/tk# PARTITIONS=$(($(fdisk -l $device | grep $device | wc -l)-1))
    root@odroidc1:/home/tk# echo $PARTITIONS
    2
    root@odroidc1:/home/tk# device="/dev/"$(lsblk -idn -o NAME | grep mmcblk0)"p"$PARTITIONS
    root@odroidc1:/home/tk# echo $device
    /dev/mmcblk0p2
    root@odroidc1:/home/tk# N=/etc/init.d/resize2fs
    root@odroidc1:/home/tk# root_device=$(mountpoint -d /)
    root@odroidc1:/home/tk# echo $root_device
    179:2
    root@odroidc1:/home/tk# for file in /dev/* ; do
    > CURRENT_DEVICE=$(printf "%d:%d" $(stat --printf="0x%t 0x%T" $file))
    > if [ $CURRENT_DEVICE = $root_device ]; then
    > root_partition=$file
    > break;
    > fi
    > done
    root@odroidc1:/home/tk# echo $root_partition
    /dev/mmcblk0p2
    root@odroidc1:/home/tk# rootfstype=$(blkid -s TYPE -o value $root_partition)
    tk@odroidc1:~$ whoami
    tk

And now I'm out. So tried to repeat this manually:

    tk@odroidc1:~$ sudo -s
    root@odroidc1:/home/tk# echo $rootfstype
    
    root@odroidc1:/home/tk# root_partition=/dev/mmcblk0p2
    root@odroidc1:/home/tk# blkid -s TYPE -o value $root_partition
    root@odroidc1:/home/tk# echo $?
    2

Since we changed that whole procedure in `firstrun` a while ago (only for `wheezy` relying on the aforementioned lenghty procedure) I tried it out with our new approach:

    root@odroidc1:/home/tk# findmnt / | awk -F" " '/\/dev\// {print $2"\t"$3}'
    /dev/mmcblk0p2	ext4

So the question is: Should I change `resize2fs` just like `firstrun` or is there something else that might fix this more easily. Also is anybody able to test this with wheezy later? Since I will definitely not build any images with this outdated distro any longer.

igorpecovnik, 2016-06-08:
I think reboot helps. Does it?

zador-blood-stained, 2016-06-08:
Last time I checked `findmnt` correctly worked on all distributions, includeing Wheezy and Trusty, both for getting block device and for FS type.

ThomasKaiser, 2016-06-08:
> I think reboot helps. Does it?

Nope, the pastebin log has been created while `/etc/init.d/resize2fs` was running (added `set -x ; exec 2>/var/log/resize2fs` at the start of the script)

I would propose that I add error handling in `firstrun` (to initiate automated reboot if necessary) and that I add our logic from `firstrun` to `resize2fs` (wheezy treating the old way and all others using `findmnt`). Or relying only on `findmnt` and creating a `motd` warning if something fails (error handling could be improved in a few more scripts)

ThomasKaiser, 2016-06-08:
It seems more that I'm dealing with Xenial. When trying to delete a partition on a device where only one is present then `fdisk` in Xenial doesn't accept the partition number:

    Command (m for help): p                                                                                                                                                 
    Disk /dev/mmcblk0: 14.5 GiB, 15523119104 bytes, 30318592 sectors                                                                                                        
    Units: sectors of 1 * 512 = 512 bytes                                                                                                                                   
    Sector size (logical/physical): 512 bytes / 512 bytes                                                                                                                   
    I/O size (minimum/optimal): 512 bytes / 512 bytes                                                                                                                       
    Disklabel type: dos                                                                                                                                                     
    Disk identifier: 0x40dc2aee                                                                                                                                             
                                                                                                                                                                            
    Device         Boot Start      End  Sectors  Size Id Type                                                                                                               
    /dev/mmcblk0p1       2048 30015391 30013344 14.3G 83 Linux                                                                                                              
                                                                                                                                                                            
    Command (m for help): d                                                                                                                                                 
    Selected partition 1                                                                                                                                                    
    Partition 1 has been deleted.                                                                                                                                           
                                                                                                                                                                            
    Command (m for help): 1                                                                                                                                                 
    1: unknown command                                                                                                                                                      

Great :)

zador-blood-stained, 2016-06-08:
It autodeletes first partition since this is the only partition available on the disk. I saw this behavior before, it probably depends more on `util-linux` version than on OS release.

ThomasKaiser, 2016-06-08:
With Xenial we get `util-linux 2.27.1` and it seems there's also some timing problem. I tried to let output redirect to `/var/log/armhwinfo.log` too and that looks both good and bad: http://sprunge.us/YeIX

I pushed my last change (that enhances the logging) but unfortunately know running completely out of time. At least it didn't worked before under same conditions so I hope I did not broke that much :\

ThomasKaiser, 2016-06-09:
I tried now to check util-linux version which seems to work: http://sprunge.us/aagA

Closed due to https://github.com/igorpecovnik/lib/commit/2e284be77d62f7516901f7657f8e773c271aa953

closed 2016-06-09 by ThomasKaiser