armbian/build #354: OrangePi Lite /dev/ttyGS* unresponsive

pctj101, 2016-06-09:
I'm trying to talk with a serial device but unable to establish communication.

Here's my GPIO readout:

```
 +-----+-----+----------+------+---+--OrangePiPC--+---+------+---------+-----+--+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 |     |     |     3.3v |      |   |  1 || 2  |   |      | 5v       |     |     |
 |   2 |   8 |    SDA.0 | ALT5 | 0 |  3 || 4  |   |      | 5V       |     |     |
 |   3 |   9 |    SCL.0 | ALT5 | 0 |  5 || 6  |   |      | 0v       |     |     |
 |   4 |   7 |   GPIO.7 | ALT3 | 0 |  7 || 8  | 0 | ALT3 | TxD3     | 15  | 14  |
 |     |     |       0v |      |   |  9 || 10 | 0 | ALT3 | RxD3     | 16  | 15  |
 |  17 |   0 |     RxD2 | ALT3 | 0 | 11 || 12 | 0 | ALT3 | GPIO.1   | 1   | 18  |
 |  27 |   2 |     TxD2 | ALT3 | 0 | 13 || 14 |   |      | 0v       |     |     |
 |  22 |   3 |     CTS2 | ALT3 | 0 | 15 || 16 | 0 | ALT3 | GPIO.4   | 4   | 23  |
 |     |     |     3.3v |      |   | 17 || 18 | 0 | ALT3 | GPIO.5   | 5   | 24  |
 |  10 |  12 |     MOSI | ALT4 | 0 | 19 || 20 |   |      | 0v       |     |     |
 |   9 |  13 |     MISO | ALT4 | 0 | 21 || 22 | 0 | ALT3 | RTS2     | 6   | 25  |
 |  11 |  14 |     SCLK | ALT4 | 0 | 23 || 24 | 0 | ALT4 | CE0      | 10  | 8   |
 |     |     |       0v |      |   | 25 || 26 | 0 | ALT3 | GPIO.11  | 11  | 7   |
 |   0 |  30 |    SDA.1 | ALT4 | 0 | 27 || 28 | 0 | ALT4 | SCL.1    | 31  | 1   |
 |   5 |  21 |  GPIO.21 | ALT3 | 0 | 29 || 30 |   |      | 0v       |     |     |
 |   6 |  22 |  GPIO.22 | ALT3 | 0 | 31 || 32 | 0 | ALT3 | RTS1     | 26  | 12  |
 |  13 |  23 |  GPIO.23 | ALT3 | 0 | 33 || 34 |   |      | 0v       |     |     |
 |  19 |  24 |  GPIO.24 | ALT3 | 0 | 35 || 36 | 0 | ALT3 | CTS1     | 27  | 16  |
 |  26 |  25 |  GPIO.25 | ALT3 | 0 | 37 || 38 | 0 | ALT3 | TxD1     | 28  | 20  |
 |     |     |       0v |      |   | 39 || 40 | 0 | ALT3 | RxD1     | 29  | 21  |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+--OrangePIPC--+------+----------+-----+-----+
```

I'm curious, is ALT3 correct?  Maybe I need a different ALT setting to actually enable /dev/ttyGS3 for TxD3 and RxD3?

pctj101, 2016-06-09:
Just to give a few more shots, I've hooked up a dumb terminal and tried all of /dev/ttyGS[0-3] with Tx/Rx normal, and swapped backwards just in case.  No luck.

zador-blood-stained, 2016-06-09:
> /dev/ttyGS* unresponsive

`/dev/ttyGSx` are usb gadget serial devices if I remember correctly, you probably want to use `/dev/ttySx`

> I'm curious, is ALT3 correct? 

According to H3 datasheet, UART1 on port bank G and UART2 on port bank A are pinmux function 2, UART3 on port bank A is pinmux function 3, so this should be correct.

I'm assuming you are using legacy kernel, so did you enable uart ports in [script.fex](https://github.com/igorpecovnik/lib/blob/master/config/fex/orangepilite.fex#L166-L191)?

pctj101, 2016-06-09:
Tried both physical Tx/RxD3 and Tx/RxD2 - with logical /dev/ttyGS* - Tx/Rx normal and backwards - no luck

pctj101, 2016-06-09:
Ah okay /dev/ttyS0 leads to the debug header - works great at boot @ 115200 baud! 

But I'm trying to use the 2x20 pin header to talk to peripherals.

pctj101, 2016-06-09:
I am using legacy kernel.

I did NOT enable uart ports in script.fex... let me see if I can figure out how to do that.

zador-blood-stained, 2016-06-09:
> Ah okay /dev/ttyS0 leads to the debug header - works great at boot @ 115200 baud!

I know 😄 

> I did NOT enable uart ports in script.fex... let me see if I can figure out how to do that.

Use `bin2fex` on your `/boot/script.bin` to get fex file, set `uart_used = 1` for ports that you want to enable, and use `fex2bin` to compile edited fex file back to `/boot/script.bin`

pctj101, 2016-06-09:
going for broke! hold on!

pctj101, 2016-06-09:
in case others are curious in the future
```
[uart3]
uart_used = 1 <- setting this
uart_port = 3
uart_type = 4
uart_tx = port:PA13<3><1><default><default>
uart_rx = port:PA14<3><1><default><default>
uart_rts = port:PA15<3><1><default><default>
uart_cts = port:PA16<3><1><default><default>
```

pctj101, 2016-06-09:
Just FYI, bin2fex -> modified as above -> fex2bin -> reboot 

same issue (FYI using physical pin 8, 10)

pctj101, 2016-06-09:
Trying with uart_type = 2 since RTS/CTS isn't really exposed on OPiLite

pctj101, 2016-06-09:
Interestingly enough uart_type = 2 for [uart3] results in

```
 +-----+-----+----------+------+---+--OrangePiPC--+---+------+---------+-----+--+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 |   4 |   7 |   GPIO.7 | ALT3 | 0 |  7 || 8  | 0 | ALT4 | TxD3     | 15  | 14  |
 |     |     |       0v |      |   |  9 || 10 | 0 | ALT4 | RxD3     | 16  | 15  |
```

pctj101, 2016-06-09:
Removing lines
```
uart_rts = port:PA15<3><1><default><default>
uart_cts = port:PA16<3><1><default><default>
```
In case it confuses uart_type 2

pctj101, 2016-06-09:
Okay still no luck.  As usual, tried Tx/Rx forwards and backwards. (and of course, GND connected :) )

pctj101, 2016-06-09:
For reference, the same serial setup works great on BananaPi Pro.
On OrangePi Lite i2c-0 WORKS (So at least we know the header is on the right direction)

zador-blood-stained, 2016-06-09:
I just connected 2 OPi One boards, UART3 to UART3 (with `uart_type = 2`), opened screen session on both boards (`screen /dev/ttyS3 115200`) and I can send text from one board to another, so UART3 does work. In case you are trying to use WiringPi or other third party library to work with serial ports, please report issue to the library maintainers - there is nothing wrong on our (Armbian) side.

pctj101, 2016-06-09:
Okay cool!  I'll make the same setup and try to verify.

Thanks for going the extra mile!

pctj101, 2016-06-09:
Oh you know what? I totally misunderstood.

I kept on banging on /dev/ttyGS3, but I didn't realize that /dev/ttyS3 appeared after amending the fex.

Thanks for your last comment.  It made all the difference!

pctj101, 2016-06-09:
Added Docs

closed 2016-06-12 by zador-blood-stained