armbian/build #358: OrangePi Lite - Power LED takes 15 seconds to turn on

pctj101, 2016-06-11:
I was wondering if there was a "reasonable" way to get that "red led" to turn on sooner indicating a boot sequence was under way.  It seems otherwise the board looks quite dead.

Perhaps there's a way to do it in the U-Boot

```
U-Boot SPL 2016.05-armbian (May 31 2016 - 18:59:26) 
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2016.05-armbian (May 31 2016 - 18:59:26 +0200) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi One
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0

```

Or some way in the U-Boot script?

```
Found U-Boot script /boot/boot.scr
2444 bytes read in 192 ms (11.7 KiB/s)
## Executing script at 43100000
gpio: pin PL10 (gpio 298) value is 1
   Warning: value of pin is still 0
gpio: pin PG11 (gpio 203) value is 1
190 bytes read in 135 ms (1000 Bytes/s)
```

It wasn't really obvious where I could stick this in.  I don't mind trying to do this myself, but was wondering where to start looking.

pctj101, 2016-06-11:
Tried to fix: https://github.com/igorpecovnik/lib/pull/359

ThomasKaiser, 2016-06-12:
Will be fixed with 5.15 release on all H3 devices.

closed 2016-06-12 by ThomasKaiser