armbian/build #360: OrangePi One (dev kernel) stops booting on CPU0: Running at unlisted freq: 1008000 KHz

squeezewrt, 2016-06-11:
Hello!

I'm compile the OS image with DEV kernel for my Orange Pi One board. OS booting stops after this message:
> 
[    8.228023] systemd-udevd[228]: starting version 215
[    8.685655] cpufreq: cpufreq_online: CPU0: Running at unlisted freq: 1008000 KHz

At the same, OS image for Orange Pi PC (on same Orange Pi One board) fires error message:

> [   12.623775] systemd-udevd[229]: starting version 215
[   14.196327] ------------[ cut here ]------------
[   14.201049] WARNING: CPU: 0 PID: 253 at drivers/clk/clk-divider.c:130 divider_recalc_rate+0x90/0xa4
[   14.210216] ths: Zero divisor and CLK_DIVIDER_ALLOW_ZERO not set
.....
[   14.412513] ---[ end trace b37dd0d879991387 ]---
[   14.417256] thermal thermal_zone0: failed to read out thermal zone (-22)

but continues booting to a successful conclusion, after which the operating system is functioning normally, even Ethernet works :)

Both logs:
[orangepi-one-image-on-orangepi-one-board.txt](https://github.com/igorpecovnik/lib/files/310459/orangepi-one-image-on-orangepi-one-board.txt)
[orangepi-pc-image-on-orangepi-one-board.txt](https://github.com/igorpecovnik/lib/files/310460/orangepi-pc-image-on-orangepi-one-board.txt)

As I understand, the 1008000 KHz frequency is not listed in **sun8i-h3-orangepi-one.dts** but is from list in **sun8i-h3-orangepi-pc.dts**  Perhaps the reason is that the Pi One and Pi PC use different frequency synthesizers to set the CPU clock, and code is running is for the wrong synthesizer?

If this is not a problem of Armbian build system, please tell me where to place issue?

I apologize for my terrible English :(
Thank You!

ThomasKaiser, 2016-06-12:
Maybe the best place is linux-sunxi IRC? Please see what I already reported back: http://irclog.whitequark.org/linux-sunxi/2016-06-09#16703354;

And https://github.com/igorpecovnik/lib/commit/224550b14adea98e5eddf36aac70cef65373d0b8 should show where to check for further adjustments. It might be possible that the few u-boot patches we're using are in conflict but as long as this THS stuff isn't confirmed to work at all priorities for an Armbian dev/next build for H3 boards (mainline kernel) are very low.

ThomasKaiser, 2016-06-12:
Hmm... I thought again about the errors we're running into and different behaviour on One/PC (which share in Armbian build system the same u-boot defconfig IIRC).

This is what megi defines for [Opi One](https://github.com/megous/linux/blob/afdf4e307e2e50b0b81a33b067141bc1c988ad89/arch/arm/boot/dts/sun8i-h3-orangepi-one.dts#L115-L120) (he took that from our Armbian fex settings and this should be adjusted anyway to our most recent values) and that for [Opi PC](https://github.com/megous/linux/blob/8c26dd23fdc69cffe3754e73c1142f11b17161e4/arch/arm/boot/dts/sun8i-h3-orangepi-pc.dts#L98-L104). Maybe it's really just a matter of bringing the different bits in sync and then THS code already works flawlessly?

But as already said: Currently very low priority (at least from my side)

squeezewrt, 2016-06-13:
OK, I close this issue.

closed 2016-06-13 by squeezewrt