armbian/build #363: [spdif bug] add a patch on a patched file

damien7851, 2016-06-13:
I think I find the kernel bug when video pause with SPDIF.
but I am not able to apply this patch because the file in question is already patched.
I have to comment on the commit that created this patch.
[https://github.com/nikkov/linux-sunxi/commit/8fb4fa09ef6defc84b15748fcb7cf93fd89efeef#diff-28e15c15c84b9a87662895772931c8c3R300](url)
how to handle this type of problem?
first issue : [https://github.com/linux-sunxi/linux-sunxi/issues/244](url)

damien7851, 2016-06-14:
I found this solution.
I run your software and stop it during kernel compilation. this produces patched sources.
I change the sources.
I commit only one modified file
I then create a patch with this commit and I put in the user patch.
it work but some patch fail because my patch include others patch.
it's easy because my patch does impact one file

igorpecovnik, 2016-06-14:
Patching order is also important, so eventually you need to add it to lib/ patches dir too see if it works properly. Try more ... and if you come out with a clean patch, try to create a pull request that we can add it.

damien7851, 2016-06-14:
I have managed to compile. the SPDIF output works but I have not tested if it crashes during a pause video I hoped to test this evening.
I am forced to make a fork to create a pull request?

damien7851, 2016-06-15:
Hi,
I have successfully tested video pause and spdif it's work.
I will make a fork and create a pull request.

closed 2016-06-15 by damien7851