armbian/build #364: can not find mali libEGL.so

svkatielee, 2016-06-14:
When I try to compile OpenCPN 4.4.0 (and earlier releases) I have to make a symbolic link to the library for the compiler to find it:
`sudo ln -s /usr/lib/mali/libEGL.so /usr/lib/`
I am compiling on Orange Pi PC with: Armbian_5.14_Orangepipc_Debian_jessie_3.4.112_desktop.raw
after an update;upgrade that needed nothing and:
 `apt-get install build-essential cmake gettext git-core gpsd gpsd-clients libgps-dev wx-common libwxgtk3.0-dev libglu1-mesa-dev libgtk2.0-dev wx3.0-headers libbz2-dev libtinyxml-dev libportaudio2 portaudio19-dev libcurl4-openssl-dev libexpat1-dev libcairo2-dev wx3.0-i18n`

ThomasKaiser, 2016-06-14:
You find the Mali BLOBs as part of [this pull request](https://github.com/linux-sunxi/sunxi-mali/pull/13) (unknown license situation)

svkatielee, 2016-06-14:
Maybe my question should have been, Is it a problem in the OS environment or in the Makefiles?

ThomasKaiser, 2016-06-14:
> Is it a problem in the OS environment or in the Makefiles?

I'm obviously not understanding the whole question. :) So do we ship with `/usr/lib/mali/libEGL.so` and the symlink is missing or what's the issue?

svkatielee, 2016-06-14:
Yes, you ship with /usr/lib/mali/libEGL.so and the symlink is missing. 
I added the symlink so the compiler could find it. I am asking if the correct action is to add the symlink in armbian or to modify the project makefiles as a special case for armbian. Should I have opened the issue at https://github.com/OpenCPN/OpenCPN

ThomasKaiser, 2016-06-14:
> I am asking if the correct action is to add the symlink in armbian or to modify the project makefiles as a special case for armbian.

Hmm... since Armbian is 'just a build system' and not a specific distro I think adding the symlink in Armbian would be the better choice. But since I'm not familiar with our desktop build at all and repackaging desktop stuff is [work in progress](http://forum.armbian.com/index.php/topic/1325-claim-a-task-set-due-date-and-start-doing-it/) we should leave this up to the new desktop/Mali maintainers.

svkatielee, 2016-06-15:
I have determined the problem is a poorly written CMakeLists.txt file cant find libs on this arch/distro.
Sorry for the noise here. Thank you for your time Thomas.

closed 2016-06-15 by svkatielee