armbian/build #368: headless Orange Pi Lite Image

holgerlembke, 2016-06-20:
Hi.

Just because I have that problem: I bought an Orange Pi Lite and can't use it. No HDMI here. Sorry. 

So why not have an image with activated totally unsecure open WIFI Access Point Mode?

zador-blood-stained, 2016-06-20:
Open Wi-FI access point doesn't sound very secure to be the default option. Maybe WPA2 with predefined password and password change prompts and reminders on login is a good idea for Lite - the only board with no Ethernet.

Meanwhile if you have [any kind of USB-UART adapter](https://linux-sunxi.org/UART) (even an Arduino board with temporarily shorted reset should work, just check the wiring - only RX on Orange is 5V tolerant), you should be able to get a shell on UART debug header and configure wireless network manually.

holgerlembke, 2016-06-20:
I know. 

My idea is to have an additional image with pointing out this security problem IN BIG RED LETTERS. Just to make it nice for hardware limited cases like me. 

I like the headless configuration of raspberry pi (via network cable). An other option would be to store SSID and PW in a simple text file and init the WIFI from there. It should be possible to change the file with Ext2Fsd and get it going.

ThomasKaiser, 2016-06-21:
And what about adding `g_serial` to `/etc/modules` on the Lite only to help with this sort of initial startup problems (and leaving it up to the user to disable/replace this module later since you mentioned [driver unloading problems](https://github.com/igorpecovnik/lib/commit/24ea1d3ea1ec0bdf823dc8fe48e4e595738af323#commitcomment-17938658))?

Providing an image with insecure Wi-Fi access is **no** option.

commit b98de7eb (2016-07-29): Enable g_serial on Orange Pi Lite by default
Closes #368
closed 2016-07-29 by zador-blood-stained