armbian/build #372: Nanopi M1 DRAM Frequency

seblucas, 2016-06-21:
Hi,

Some stuff I left behind about the deceiving Nanopi M1, I've used lima_memtester and went up to 744MHz using tkaiser test image (see [here](http://forum.armbian.com/index.php/topic/1322-testers-wanted-testing-dram-reliability-on-bpi-m2-and-nanopi-m1/page-2#entry10314)). It seems that the default frequency is 624MHz.

I still have a hard time finding my way in Armbian sources but I think I can update https://github.com/igorpecovnik/lib/blob/master/config/fex/nanopim1.fex#L109. Do I have to update something with uboot ?

Another question do we need to change the default value or just update the documentation in case a user want to try ?

ThomasKaiser, 2016-06-21:
U-boot sets DRAM clockspeed to 624 MHz for all H3 devices (and this will remain the same). The kernel uses this clockspeed from then on but with our legacy sun8i kernel code and settings (adjust DRAM freq as part of throttling) the first time the budget cooling stuff happens, DRAM clockspeed will be reduced and afterwards adjusted to the value used in the fex file. Since NanoPi M1 is 'hot stuff' simply changing this in the fex should be enough.

IMO there's is absolutely no need to document stuff like this since we should try to keep the overclocker's camp out of Armbian. If you increase DRAM clockspeed from 624MHz to 744MHz you won't notice this in any real world performance (I was only able to measure this stuff with lima-bandwidth tester or something like that: a few percent more or less regarding memcopy operations). If we document this, some people will start to do so without appropriately testing reliability using lima-memtester, then get instabilites and / or data corruption (all the nice by-products of DRAM bit flips) and will start to report that Armbian is not reliable. No thanks :)

closed 2016-06-21 by ThomasKaiser

seblucas, 2016-06-21:
Thanks for the clarification ;)