armbian/build #376: A wireless module 8192cu (EW-7811Un -Realtek RTL8188CUS ) on Orange PI PC

s-s-s, 2016-06-24:
Didn't work on OrangePi PC with Armbian_5.14_Orangepipc_Debian_jessie_3.4.112 still (
It is unable to make a connection with router although scanning for networks does work.

```

# uname -a
Linux orangepipc 3.4.112-sun8i #2 SMP PREEMPT Sun Jun 5 08:38:56 CEST 2016 armv7l GNU/Linux

```

```
# lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
Jun 23 13:01:36 localhost kernel: [    4.551979] USB_SPEED_HIGH
Jun 23 13:01:36 localhost kernel: [    4.552012] CHIP TYPE: RTL8188C_8192C
Jun 23 13:01:36 localhost kernel: [    4.552055] register rtw_netdev_ops to netdev_ops
Jun 23 13:01:36 localhost kernel: [    4.552318] Chip Version ID: VERSION_NORMAL_TSMC_CHIP_88C.
Jun 23 13:01:36 localhost kernel: [    4.552329] RF_Type is 3!!
Jun 23 13:01:36 localhost kernel: [    4.552766] EEPROM type is E-FUSE
Jun 23 13:01:36 localhost kernel: [    4.552774] ====> ReadAdapterInfo8192C
Jun 23 13:01:36 localhost kernel: [    4.552891] Boot from EFUSE, Autoload OK !
Jun 23 13:01:36 localhost kernel: [    4.703781] EEPROMVID = 0x7392
Jun 23 13:01:36 localhost kernel: [    4.703794] EEPROMPID = 0x7811
Jun 23 13:01:36 localhost kernel: [    4.703800] EEPROMCustomerID : 0x00
Jun 23 13:01:36 localhost kernel: [    4.703806] EEPROMSubCustomerID: 0x00
Jun 23 13:01:36 localhost kernel: [    4.703812] RT_CustomerID: 0x00
Jun 23 13:01:36 localhost kernel: [    4.703823] _ReadMACAddress MAC Address from EFUSE = 74:da:38:41:69:6a
Jun 23 13:01:36 localhost kernel: [    4.703834] EEPROMRegulatory = 0x0
Jun 23 13:01:36 localhost kernel: [    4.703840] _ReadBoardType(0)
Jun 23 13:01:36 localhost kernel: [    4.703847] BT Coexistance = disable
Jun 23 13:01:36 localhost kernel: [    4.703854] mlmepriv.ChannelPlan = 0x02
Jun 23 13:01:36 localhost kernel: [    4.703863] _ReadPSSetting...bHWPwrPindetect(0)-bHWPowerdown(0) ,bSupportRemoteWakeup(0)
Jun 23 13:01:36 localhost kernel: [    4.703874] ### PS params=>  power_mgnt(1),usbss_enable(0) ###
Jun 23 13:01:36 localhost kernel: [    4.703881] ### AntDivCfg(0)
Jun 23 13:01:36 localhost kernel: [    4.703887] readAdapterInfo_8192CU(): REPLACEMENT = 1
Jun 23 13:01:36 localhost kernel: [    4.703898] <==== ReadAdapterInfo8192C in 150 ms
Jun 23 13:01:36 localhost kernel: [    4.704627] rtw_macaddr_cfg MAC Address  = 74:da:38:41:69:6a
Jun 23 13:01:36 localhost kernel: [    4.704643] bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
Jun 23 13:01:36 localhost kernel: [    4.705581] _rtw_drv_register_netdev, MAC Address (if1) = 74:da:38:41:69:6a
Jun 23 13:01:36 localhost kernel: [    4.705708] usbcore: registered new interface driver rtl8192cu
Jun 23 13:01:36 localhost kernel: [   70.438889] +871x_drv - drv_open, bup=0
Jun 23 13:01:36 localhost kernel: [   70.443159]  ===> FirmwareDownload91C() fw:Rtl819XFwImageArray_TSMC
Jun 23 13:01:36 localhost kernel: [   70.443171] FirmwareDownload92C accquire FW from embedded image
Jun 23 13:01:36 localhost kernel: [   70.443180] fw_ver=v88, fw_subver=2, sig=0x88c0
Jun 23 13:01:36 localhost kernel: [   70.468396] fw download ok!
Jun 23 13:01:36 localhost kernel: [   70.468404] Set RF Chip ID to RF_6052 and RF type to 1T1R.
Jun 23 13:01:36 localhost kernel: [   70.857050] IQK:Start!!!
Jun 23 13:01:36 localhost kernel: [   70.867923] Path A IQK Success!!
Jun 23 13:01:36 localhost kernel: [   70.875673] Path A IQK Success!!
Jun 23 13:01:36 localhost kernel: [   70.880302] IQK: final_candidate is 0
Jun 23 13:01:36 localhost kernel: [   70.880331] IQK: RegE94=103 RegE9C=11 RegEA4=ff RegEAC=3ff RegEB4=0 RegEBC=0 RegEC4=0 RegECC=0
Jun 23 13:01:36 localhost kernel: [   70.880346]  Path A IQ Calibration Success !
Jun 23 13:01:36 localhost kernel: [   70.993928] pdmpriv->TxPowerTrackControl = 1
Jun 23 13:01:36 localhost kernel: [   70.998552] rtl8192cu_hal_init in 560ms
Jun 23 13:01:36 localhost kernel: [   71.015560] MAC Address = 74:da:38:41:69:6a
Jun 23 13:01:36 localhost kernel: [   71.015988] -871x_drv - drv_open, bup=1
Jun 23 13:01:36 localhost kernel: [   71.017693] ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jun 23 13:01:36 localhost kernel: [   71.017790] [rtw_wx_set_pmkid] IW_PMKSA_FLUSH!
Jun 23 13:01:36 localhost kernel: [   71.017830] set_mode = IW_MODE_INFRA
[... 37 lines cut ...]
Jun 23 13:01:36 localhost kernel: [   98.542167] IQK: final_candidate is 0
Jun 23 13:01:36 localhost kernel: [   98.542195] IQK: RegE94=103 RegE9C=10 RegEA4=ff RegEAC=3ff RegEB4=0 RegEBC=0 RegEC4=0 RegECC=0
Jun 23 13:01:36 localhost kernel: [   98.542210]  Path A IQ Calibration Success !
Jun 23 13:01:36 localhost kernel: [   98.653927] pdmpriv->TxPowerTrackControl = 1
Jun 23 13:01:36 localhost kernel: [   98.658677] rtl8192cu_hal_init in 580ms
Jun 23 13:01:36 localhost kernel: [   98.675742] <===  rtw_ips_pwr_up.............. in 600ms
Jun 23 13:01:36 localhost kernel: [   98.675929] ==> ips_leave.....LED(0x00028080)...
Jun 23 13:01:36 localhost kernel: [   99.986056] survey done event(22) band:0 for wlan0
Jun 23 13:01:36 localhost kernel: [  102.670104] ==>rtw_ps_processor .fw_state(8)
Jun 23 13:01:36 localhost kernel: [  102.670129] ==>ips_enter cnts:5
Jun 23 13:01:36 localhost kernel: [  102.670144] ===> rtw_ips_pwr_down...................
Jun 23 13:01:36 localhost kernel: [  102.670425] ====> rtw_ips_dev_unload...
Jun 23 13:01:36 localhost kernel: [  102.693927] usb_read_port_cancel
Jun 23 13:01:36 localhost kernel: [  102.694023] usb_read_port_complete()-1284: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0) bReadPortCancel(1)
Jun 23 13:01:36 localhost kernel: [  102.694144] usb_read_port_complete()-1284: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0) bReadPortCancel(1)
Jun 23 13:01:36 localhost kernel: [  102.694268] usb_read_port_complete()-1284: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0) bReadPortCancel(1)
Jun 23 13:01:36 localhost kernel: [  102.694392] usb_read_port_complete()-1284: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0) bReadPortCancel(1)
Jun 23 13:01:36 localhost kernel: [  102.694447] usb_write_port_cancel
Jun 23 13:01:36 localhost kernel: [  102.694534] ==> rtl8192cu_hal_deinit
Jun 23 13:01:36 localhost kernel: [  102.694550] bkeepfwalive(0)
Jun 23 13:01:36 localhost kernel: [  102.694564] card disble without HWSM...........
Jun 23 13:01:36 localhost kernel: [  102.699676] <=== rtw_ips_pwr_down..................... in 20ms
Jun 23 13:01:36 localhost kernel: [  104.993781] _rtw_pwr_wakeup call ips_leave....
Jun 23 13:01:36 localhost kernel: [  104.993810] ==>ips_leave cnts:5
```

igorpecovnik, 2016-06-24:
We try to support on board wifi chips, the rest is far too much for us. This is general Linux, bad chip vendor, driver issue. You have to search net or ask chip manufacturers for better drivers. We prepared a kernel that you can add drivers on your own: https://github.com/igorpecovnik/lib/blob/master/documentation/user-faq.md#how-to-build-a-wireless-driver- (perhaps even the driver from example will work)

If you succeed to find a better working driver, please report and than we can invest up to few hours to bring this (better) working driver.

Not our issue.

closed 2016-06-24 by igorpecovnik

s-s-s, 2016-06-24:
Ok. I'll try.
Thank you for answer.

s-s-s, 2016-06-24:
After compiling the three versions of the drivers the problem was resolved by the transfer of the access point in the router in mode "Do not hide SSID" :)