armbian/build #377: uInitrd

sgufa, 2016-06-24:
i've noticed that after build process i cant find /boot/uInitrd. Seems that something goes wrong converting initrd.img to uInitrd

zador-blood-stained, 2016-06-24:
Build process for which board and kernel exactly?

sgufa, 2016-06-24:
cubieboard2 wheezy legacy console only

zador-blood-stained, 2016-06-24:
I just built an image - cubieboard2, legacy, wheezy, CLI
```
# ls /boot
bin       config-3.4.112-sun7i      uInitrd
boot.bmp  initrd.img-3.4.112-sun7i  vmlinuz-3.4.112-sun7i
boot.cmd  script.bin                zImage
boot.scr  System.map-3.4.112-sun7i
```

ThomasKaiser, 2016-06-24:
@sgufa What I sometimes do when 'strange things' happen:

    mkdir ../test ; cd ../test
    git clone --depth 1 https://github.com/igorpecovnik/lib
    ...

(and then let the same build run unattended and do something different in the meantime). Just to check whether a clean git checkout shows also problems or not so that I can then decide to start investigating or working on the new checkout and trashing the old one (depends on my priorities then)

sgufa, 2016-06-24:
eheh...will give a try

sgufa, 2016-06-27:
so i've tried in a clean situation with the latest build environment. Same issue

zador-blood-stained, 2016-06-27:
Please set
```
PROGRESS_LOG_TO_FILE=yes
```
in compile.sh, delete all files in `output/cache/rootfs`, launch build process again and provide output.log  and debootstrap.log from `output/debug` directory.

sgufa, 2016-06-27:
[dbg.zip](https://github.com/igorpecovnik/lib/files/334639/dbg.zip)

zador-blood-stained, 2016-06-27:
So it's not Wheezy but Trusty, and now the problem can be reproduced.

sgufa, 2016-06-27:
oops... :)

zador-blood-stained, 2016-06-27:
Should work now

sgufa, 2016-06-27:
yeah! it works

closed 2016-06-27 by sgufa