armbian/build #379: nand-sata-install has hard-coded mmcblk1

martinayotte, 2016-06-25:
Under OrangePi-Plus2E, the eMMC is appearing under /dev/mmcblk2.
I had to tweak /usr/sbin/nand-sata-install to change all references to /dev/mmcblk1 to /dev/mmcblk2 manually to get the script working.

ThomasKaiser, 2016-06-25:
Please be careful: `/usr/sbin/nand-sata-install` works flawlessly with legacy kernel (I run both OPi Plus 2E and PC Plus from eMMC) and there eMMC is still /dev/mmcblk1. Better investigate why the device node is called differently with mainline kernel (SDIO node for Wi-Fi in between?) 

I remember reading some stuff from wens and Hans about mainline, SDIO and DT but got lost...

igorpecovnik, 2016-06-25:
Agree and I think this could be some general issue? I notice similar problem on imx boards but it was fixed on one update so I didn't investigate further. There, SD card become mmcblk1.

martinayotte, 2016-06-25:
I think the chosen numbering in Mainline for mmc1 as SDIO and mmc2 for eMMC is simply that you can have an board with the Wifi without any eMMC, such OrangePi-Lite, but we won't find any OrangePi with an eMMC without Wifi. 
Since those pins were already assigned by Hans, I would not questioned it too much. ;-)

martinayotte, 2016-06-25:
BTW, on a side note, it seems that I faced another issue with "nand-sata-install" where it didn't update the boot.cmd during the install on /dev/mmcblk2, the value was still "root=' /dev/mmcblk0'" after installation, but I wasn't aware until I've spend some time to debug that out.

igorpecovnik, 2016-08-06:
This should be fixed with:
https://github.com/igorpecovnik/lib/commit/c1808c85a38d320c77c102b126c8809138f5d113

closed 2016-08-06 by igorpecovnik