armbian/build #383: Unsupported ioctl: cmd=0x8933 with Ansible within the chroot

fheslouin, 2016-06-27:
I'm using the last version of lib. I added a wget in customize-image.sh to download an Ansible script to execute during installation. 

I'm getting this issue : ioctl 0x8933

The Ansible script works well on any other machine. It might be related to this https://gist.github.com/tpokorra/8bc85d2963ddc219ba0a

```
Unsupported ioctl: cmd=0x8933                                                                                                                                                             [14/945]
localhost | SUCCESS => {
    "after": "8cb9e92a7d9ad6684bcf52126101f7c8f9e5a347", 
    "before": "f34a90bfd90f6ee3c6c36a95e55b7696b15a8492", 
    "changed": true
}

PLAY [localhost] ***************************************************************

TASK [setup] *******************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "failed": true, "module_stderr": "Unsupported ioctl: cmd=0x8933\nUnsupported ioctl: cmd=0x8933\nTraceback (most recent call last):\n  File \"/tm
p/ansible_r7eg4g/ansible_module_setup.py\", line 127, in <module>\n    main()\n  File \"/tmp/ansible_r7eg4g/ansible_module_setup.py\", line 119, in main\n    data = get_all_facts(module)\n  File
 \"/tmp/ansible_r7eg4g/ansible_modlib.zip/ansible/module_utils/facts.py\", line 3177, in get_all_facts\n  File \"/tmp/ansible_r7eg4g/ansible_modlib.zip/ansible/module_utils/facts.py\", line 3123
, in ansible_facts\n  File \"/tmp/ansible_r7eg4g/ansible_modlib.zip/ansible/module_utils/facts.py\", line 919, in populate\n  File \"/tmp/ansible_r7eg4g/ansible_modlib.zip/ansible/module_utils/f
acts.py\", line 107, in wrapper\n  File \"/tmp/ansible_r7eg4g/ansible_modlib.zip/ansible/module_utils/facts.py\", line 1142, in get_mount_facts\nIndexError: list index out of range\n", "module_s
tdout": "", "msg": "MODULE FAILURE", "parsed": false}

NO MORE HOSTS LEFT *************************************************************
        to retry, use: --limit @/var/lib/ansible/local/main.retry
```

zador-blood-stained, 2016-06-27:
If QEMU doesn't support some syscalls, you can't do anything about it. If you are not using Ubuntu Xenial as build host, please try using it (or install qemu-user-static package from Xenial). If it doesn't help, then this means that you can't run this script in chroot.

fheslouin, 2016-06-27:
Thanks, I'm using trusty right now, I'm gonna try with Xenial and let you know in this post if it solved the issue.

lanefu, 2016-06-27:
@fheslouin 

It looks like the ansible setup module is failing on checking the mounts--seems chroot related.   If your ansible version is the latest you could try filtering the fact it gathers and skip the mounts...

http://docs.ansible.com/ansible/setup_module.html

Or if your playbook doesn't need local facts at all

`gather_facts: False`

fheslouin, 2016-06-28:
The last ubuntu version (Xenial) fixed the ioctl issue. 

As @lanefu said their is probably a problem around ansible trying to gather facts from the host and the mounted devices. 
`gather_facts: False` does not solve the problem but the guys from Ansible says that `/etc/mtab` could be missing in the chroot.
I tried to look for it and indeed I was not able to find it...

Is their any way to solve this issue ?

zador-blood-stained, 2016-06-28:
> the guys from Ansible says that /etc/mtab could be missing in the chroot.

`/etc/mtab` should be symlinked to `/proc/self/mounts`. You may try removing this symlink, creating empty file (or with some dummy contents) before running installation script, and restoring this symlink after your installation script.

fheslouin, 2016-06-28:
Great news ! Adding a fake `/etc/mtab` did the trick !
Ansible was looking for this file in order to work correctly.

Thanks for the help, I close the issue.

closed 2016-06-28 by fheslouin