armbian/build #385: [opipcplus] Problem with compilation in Docker container

pietrushnic, 2016-06-29:
Compilation of OragePi PC Plus in Docker container fails:

```
sudo docker build -t lib .
sudo docker run -it -P --privileged lib
./compile.sh BOARD=orangepipcplus PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=dev APT_PROXY_ADDR=<ip_addr>:3142
```

For `default` I have very similar issues.

Build log:

```
(...)
patching file tools/include/tools/be_byteshift.h
patching file tools/include/tools/le_byteshift.h
  CLEAN   scripts/basic
  CLEAN   scripts/dtc
  CLEAN   scripts/kconfig
  CLEAN   scripts/mod
  CLEAN   scripts
dpkg-deb: building package 'linux-firmware-image-dev-sun8i' in '../linux-firmware-image-dev-sun8i_5.14_armhf.deb'.
dpkg-deb: building package 'linux-dtb-dev-sun8i' in '../linux-dtb-dev-sun8i_5.14_armhf.deb'.
dpkg-deb: building package 'linux-headers-dev-sun8i' in '../linux-headers-dev-sun8i_5.14_armhf.deb'.
dpkg-deb: building package 'linux-image-dev-sun8i' in '../linux-image-dev-sun8i_5.14_armhf.deb'.
dpkg-genchanges: warning: package linux-libc-dev-dev-sun8i in control file but not in files list
dpkg-genchanges: binary-only upload (no source code included)
[ o.k. ] Creating board support package [ orangepipcplus dev ]
[ o.k. ] Fingerprinting [ Armbian 5.14 Orangepipcplus Debian jessie dev ]
[ o.k. ] Building package [ linux-jessie-root-dev-orangepipcplus ]
[ o.k. ] Starting build process for [ orangepipcplus jessie ]
[ o.k. ] Extracting jessie-ng-armhf.094...b9d.tgz [ 0 days old ]
jessie-ng-armhf.094...b9d.tgz: 99.3MiB [ 158MiB/s] [========================================================================================================================================================>] 100%
[ o.k. ] Applying distribution specific tweaks for [ jessie ]
sed: can't read /root/output/cache/sdcard/etc/ssh/sshd_config: No such file or directory
chroot: failed to run command '/bin/bash': Exec format error
[ o.k. ] Applying common tweaks 
rm: cannot remove '/root/output/cache/sdcard/etc/network/interfaces': No such file or directory
sed: can't read /root/output/cache/sdcard/etc/default/console-setup: No such file or directory
chroot: failed to run command '/bin/bash': Exec format error
chroot: failed to run command '/bin/bash': Exec format error
chroot: failed to run command '/bin/bash': Exec format error
[ o.k. ] Installing kernel [ linux-image-dev-sun8i ]
[ o.k. ] Installing u-boot [ linux-u-boot-dev-orangepipcplus ]
[ o.k. ] Installing headers [ linux-headers-dev-sun8i ]
[ o.k. ] Installing DTB [ linux-dtb-dev-sun8i ]
[ o.k. ] Installing board support package [ orangepipcplus ]
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
cp: cannot create regular file '/root/output/cache/sdcard/etc/lirc/lircd.conf': No such file or directory
chroot: failed to run command '/bin/bash': Exec format error
[ o.k. ] Installing extra applications and drivers 
[ o.k. ] Installing linux firmware [ 5.14 ]
chroot: failed to run command '/bin/bash': Exec format error
[ o.k. ] Building deb [ hostapd ]
[ o.k. ] Downloading sources [ hostapd ]
[ o.k. ] Compiling hostapd [ v2.5 ]
[ error ] ERROR in function compiling [ hostapd.sh:72 ]
[ error ] Error building [ hostapd ]
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:553 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

Any idea what can be wrong ?

igorpecovnik, 2016-06-29:
rootfs cache file is too small, presumably corrupted. Remove output/cache/rootfs and run again.

lanefu, 2016-06-29:
@pietrushnic 

I advise against doing image builds in docker.  Docker containers are fine for doing armbian kernel builds, but problematic for image builds.

The armbian image build process relies on creating loopback image mounts.    I spent a day and a half troubleshooting the process... Ultimately getting the loopback kernel modules to create proper loopback devices within the container repeatedly proved to be unreliable.

(I was using CentOS7 as my docker host with a ubuntu 14.04 Docker container for the build)

zador-blood-stained, 2016-06-29:
> Your error is very familar to me. If you take a look at your container for another share you'll probably see that the sdcard folder is not mounted to a loopback to device.

```
--privileged=true|false 
Give extended privileges to this container. 
By default, Docker containers are “unprivileged” (=false) and cannot, for example, run a Docker 
daemon inside the Docker container. This is because by default a container is not allowed to 
access any devices. A “privileged” container is given access to all devices.
```

loop device is not required at rootfs cache creation stage, so it's probably an `apt-cacher` issue again **or** issue with package list - hash in file name `jessie-ng-armhf.094...b9d.tgz` doesn't match current value for Jessie CLI.

pietrushnic, 2016-06-29:
@lanefu unfortunately I think avoiding using containers is not good way to go. Containers should work mainly because people that do many builds want to have separate embedded development environment. I hate idea of installing something in my system because Armbian requires that. I do a lot of builds from scratch, buidroot and  always try to use contrainers because it give me isolation. I'm willing to help as much as I can to make contrainers work fine with Armbian.

@zador-blood-stained  not sure about `apt-cacher` reference, we resolve apt proxy before and I did the check about that before build (`<ip_addr>:3142` show required content) and it seems to be installed and working. `--privilidged` option was advised in #255. Not matching hash is for sure bad indicator. I tried to build on machine with 8GB of RAM, no retry with 16GB of RAM. Maybe method presented in #255 is not the best for this use case ?

zador-blood-stained, 2016-06-29:
@pietrushnic 
"Exec format error" may indicate that QEMU in chroot doesn't work properly, but this needs to be investigated in steps.

So, please, delete rootfs cache and run building again. If there will be any errors, please post log files from `output/debug` directory and full console output..
Second, what is your host OS? QEMU/binfmt needs kernel support AFAIK, so it may require additional steps or parameters to make it work...

pietrushnic, 2016-06-29:
@zador-blood-stained my host is Debian sid. I did completely clean test and it fails on my 8GB machine, but I retried that with 16GB workstation and build passed.

I think that default procedure for building in docker should be changed and output of build should be placed on hard disk by mounted volume not kept in container. I think this issue can be closed, but there should be guide on building in container. Where I can place that ?

pietrushnic, 2016-06-29:
@zador-blood-stained I was too fast with that:

```
[ o.k. ] Ending debootstrap process and preparing cache [ jessie ]
jessie-ng-armhf.094...b9d.tgz:  645MiB [ 102MiB/s] [====================================================================================================================================================================================] 109%
[ o.k. ] Applying distribution specific tweaks for [ jessie ]
[ o.k. ] Applying common tweaks 
[ o.k. ] Installing kernel [ linux-image-dev-sun8i ]
[ o.k. ] Installing u-boot [ linux-u-boot-dev-orangepipcplus ]
[ o.k. ] Installing headers [ linux-headers-dev-sun8i ]
[ o.k. ] Installing DTB [ linux-dtb-dev-sun8i ]
[ o.k. ] Installing board support package [ orangepipcplus ]
[ o.k. ] Installing extra applications and drivers 
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Installing linux firmware [ 5.14 ]
[ o.k. ] Building deb [ hostapd ]
[ o.k. ] Downloading sources [ hostapd ]
[ o.k. ] Compiling hostapd [ v2.5 ]
[ o.k. ] Installing [ armbian-hostapd-jessie_5.14_armhf.deb ]
invoke-rc.d: policy-rc.d denied execution of stop.
invoke-rc.d: policy-rc.d denied execution of start.
[ o.k. ] Building deb [ sunxi-tools ]
[ o.k. ] ... downloading sources [ sunxi-tools ]
[ o.k. ] ... downloading sources [ temper ]
[ o.k. ] ... downloading sources [ BT utils ]
[ o.k. ] ... compiling [ sunxitools ]
[ o.k. ] ... compiling [ temper ]
[ o.k. ] ... compiling [ bluetooth utils ]
[ o.k. ] Installing [ armbian-tools-jessie_5.14_armhf.deb ]
[ o.k. ] Installing additional application [ USB redirector ]
cp: cannot stat '/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/tusbd.ko': No such file or directory
[ o.k. ] Calling image customization script [ customize-image.sh ]
[ o.k. ] Preparing image file for rootfs [ orangepipcplus jessie ]
[ o.k. ] Current rootfs size [ 832 MiB ]
[ o.k. ] Creating blank image for rootfs [ 1294 MiB ]
1.26GiB [ 239MiB/s] [==================================================================================================================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
sh: 1: udevadm: not found
sh: 1: udevadm: not found
sh: 1: udevadm: not found
sh: 1: udevadm: not found
sh: 1: udevadm: not found
sh: 1: udevadm: not found
The file /dev/loop2p1 does not exist and no size was specified.
tune2fs: No such file or directory while trying to open /dev/loop2p1
Couldn't find valid filesystem superblock.
mount: special device /dev/loop2p1 does not exist
[ o.k. ] Copying files to image [ tmprootfs.raw ]
        753.44M  99%  222.93MB/s    0:00:03 (xfr#41218, to-chk=0/52030)   

sent 756.51M bytes  received 829.95K bytes  216.38M bytes/sec
total size is 757.35M  speedup is 1.00
[ o.k. ] Copying files to /boot partition [ tmprootfs.raw ]
         19.17M  99%  168.97MB/s    0:00:00 (xfr#139, to-chk=0/145) 

sent 19.18M bytes  received 2.67K bytes  38.37M bytes/sec
total size is 19.17M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
tmpfs                                                                                              1.5G  832M  669M  56% /root/output/cache/sdcard
[ o.k. ] Writing bootloader [ /dev/loop2 ]
umount: /root/output/cache/mount: not mounted
[ o.k. ] Copying image file [ Armbian_5.14_Orangepipcplus_Debian_jessie_4.6.2.raw ]
[ o.k. ] Done building [ /root/output/images/Armbian_5.14_Orangepipcplus_Debian_jessie_4.6.2.raw ]
[ o.k. ] Runtime [ 30 min ]
```
Build finishes fine, but it seem to not be correct after compressing it I have 1.5MB `tar.gz`. `md5sum` is exactly the same as in container. I will check that if it boots tomorrow since I left board in office.

pietrushnic, 2016-06-29:
`logs-29_06_2016-21_44_21.tar.gz` doesn't see to contain correct content. Logs:

* [compilation.log](http://paste.ubuntu.com/18131383/) - it has some warnings
* [debootstrap.log](http://paste.ubuntu.com/18131392/)
* [hostapd-build.log](http://paste.ubuntu.com/18131400/)
* [install.log](http://paste.ubuntu.com/18131408/)
* [output.log](http://paste.ubuntu.com/18131417/)
* [patching.log](http://paste.ubuntu.com/18131422/)
* [tools-build.log](http://paste.ubuntu.com/18131430/)

lanefu, 2016-06-29:
@pietrushnic Your hitting the loopback device problem.. it's right there in your console output:

> The file /dev/loop2p1 does not exist and no size was specified.
> tune2fs: No such file or directory while trying to open /dev/loop2p1
> Couldn't find valid filesystem superblock.
> mount: special device [...]

It's a critical step.  Its failure makes your image incomplete.  You have to make the loopback mounts work.

Ideologically, I agree with you.. containers _should_ work for building images.  The reality is that once you've pushed your container to requiring privileged mode, the _deploy anywhere_ paradigm of docker containers is lost.   The constraints become the capability of the docker host.  Repeatability is lost.  You won't be able to use it in any container-only hosting service such as on AWS or google compute engine.   

I do admire your desire to overcome.  Loopback is the hurdle if you want to tackle it.  If you do some searching you'll see others struggling with loopback devices inside of containers.  Its not a unique scenario.

See my post [here]( http://forum.armbian.com/index.php/topic/449-testing-running-armbian-tools-with-docker-style-vm/?p=9458) for a few tricks to improve your luck with building in a container.  

Please share your technique if you come up with a repeatable solution.

zador-blood-stained, 2016-06-29:
Please try new branch "docker-test" - you may have to pull and check out it manually and lock changes by creating file `.ignore_changes` near compile.sh, or setting LIB_TAG="docker-test" in compile.sh may be enough.

pietrushnic, 2016-06-30:
@zador-blood-stained I followed `LIB_TAG="docker-test"`. Result is much better final image has some content (`tar.gz` is ~300MB), but there is still issue in `install.log`:

```
In file included from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/usbd.h:67:0,
                 from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:13:
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c: In function 'IIIIIIllI':
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/utils.h:77:83: error: inlining failed in call to always_inline 'IIlIIIlII': function body not available
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:197:18: error: called from here
 ==BUS_NOTIFY_ADD_DEVICE){if(IIlIIIlII(lIIIIl)){struct usb_device*lIIIl=
                  ^
In file included from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/usbd.h:67:0,
                 from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:13:
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/utils.h:77:194: error: inlining failed in call to always_inline 'IIllIlIIl': function body not available
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:215:12: error: called from here
 }}}else if(IIllIlIIl(lIIIIl)){struct usb_interface*llIIl=to_usb_interface(dev);
            ^
In file included from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/usbd.h:67:0,
                 from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:13:
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/utils.h:77:194: error: inlining failed in call to always_inline 'IIllIlIIl': function body not available
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:269:59: error: called from here
 ,lIIIII(&llIIl->dev));}}}}else if(IIllIII==BUS_NOTIFY_DEL_DEVICE){if(IIllIlIIl(
                                                           ^
In file included from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/usbd.h:67:0,
                 from /root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:13:
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/utils.h:77:83: error: inlining failed in call to always_inline 'IIlIIIlII': function body not available
/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.c:275:11: error: called from here
 }}else if(IIlIIIlII(lIIIIl)){struct usb_device*lIIIl=to_usb_device(lIIIIl);if(
           ^
scripts/Makefile.build:291: recipe for target '/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.o' failed
make[2]: *** [/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/loader.o] Error 1
Makefile:1429: recipe for target '_module_/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd' failed
make[1]: *** [_module_/root/sources/usb-redirector-linux-arm-eabi/files/modules/src/tusbd] Error 2
make[1]: Leaving directory '/root/sources/linux-sun8i-mainline/orange-pi-4.6'
Makefile:61: recipe for target 'default' failed
make: *** [default] Error 2
```

Looks like `usb-redirector` has some problems with compilation.

I will try image tomorrow and let you know.

zador-blood-stained, 2016-06-30:
> I did completely clean test and it fails on my 8GB machine, but I retried that with 16GB workstation and build passed.

If there still are errors when building on any HW configuration, I would like to see the logs and full console output. It should not depend on available RAM size in any case.

> Looks like usb-redirector has some problems with compilation.

It may be broken for some configurations, but it's not essential and it shouldn't affect anything.

pietrushnic, 2016-06-30:
@zador-blood-stained it still fails on my 8GB Debian sid laptop:

http://paste.ubuntu.com/18166338/
http://paste.ubuntu.com/18166340/
http://paste.ubuntu.com/18166341/
http://paste.ubuntu.com/18166345/
http://paste.ubuntu.com/18166346/

Image build from your `docker-test` branch on my 16GB workstation works fine so far.

I tried to change approach today by using volume and local repository:

```
docker run -v ${PWD}/lib:/root/lib -v ${PWD}/output:/root/output -it -P --privileged=true lib
```

Build command:

```
./compile.sh BOARD=orangepipcplus PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=dev APT_PROXY_ADDR=<ip_addr>:3142 LIB_TAG="docker-test" CLEAN_CACHE="cache"
```

zador-blood-stained, 2016-06-30:
> CLEAN_CACHE="cache"

it should be `CLEAN_LEVEL=cache`. You used previously built rootfs cache and debootstrap log is still missing.

> it still fails on my 8GB Debian sid laptop

Do you use Docker and `docker-test` branch here too?

pietrushnic, 2016-06-30:
There is no `debootstrap.log`:

```
root@6f6992e9d113:~# ll output/debug/
total 600
drwxr-xr-x 2 root root   4096 Jun 30 12:21 ./
drwxr-xr-x 6 1000 1000   4096 Jun 30 12:05 ../
-rw-r--r-- 1 root root 475233 Jun 30 12:20 compilation.log
-rw-r--r-- 1 root root   1591 Jun 30 12:21 hostapd-build.log
-rw-r--r-- 1 root root   3287 Jun 30 12:20 install.log
-rw-r--r-- 1 root root     45 Jun 30 11:49 logs-30_06_2016-11_49_51.tgz
-rw-r--r-- 1 root root  89755 Jun 30 12:17 logs-30_06_2016-12_17_19.tgz
-rw-r--r-- 1 root root   9969 Jun 30 12:17 logs-30_06_2016-12_17_54.tgz
-rw-r--r-- 1 root root   5118 Jun 30 12:21 output.log
-rw-r--r-- 1 root root     42 Jun 30 12:18 patching.log
```

I will try forcing `docker-test` through `./compile.sh`.

In case of `CLEAN_CACHE` documentation should be changed: https://github.com/igorpecovnik/lib/wiki/Build-options

zador-blood-stained, 2016-06-30:
> In case of CLEAN_CACHE documentation should be changed: https://github.com/igorpecovnik/lib/wiki/Build-options

It definitely says `CLEAN_LEVEL` there, not `CLEAN_CACHE`

`debootstrap.log` is created if you don't have rootfs cache

pietrushnic, 2016-06-30:
@zador-blood-stained ok, sorry, I thought more about `CLEAN_LEVEL=cache` vs `CLEAN_LEVEL="cache"` but assume both should work. Rebuilding with correct cleaning flag and `docker-test` branch.

faddat, 2016-07-02:
I've got to add that I've found that LXC works better, but if you're not on ubuntu, (arch?) systemd-nspawn seemed to work best for me in the end.  Unforunatley docker containers are subject to all of the wierdness of docker containers :/.

kometchtech, 2016-07-04:
I tried to follow, at Docker can not build an image.

http://forum.armbian.com/index.php/topic/449-testing-running-armbian-tools-with-docker-style-vm/?hl=docker

```
dpkg-deb: building package 'linux-firmware-image-dev-sun8i' in '../linux-firmware-image-dev-sun8i_5.16_armhf.deb'.
dpkg-deb: building package 'linux-dtb-dev-sun8i' in '../linux-dtb-dev-sun8i_5.16_armhf.deb'.
dpkg-deb: building package 'linux-headers-dev-sun8i' in '../linux-headers-dev-sun8i_5.16_armhf.deb'.
dpkg-deb: building package 'linux-image-dev-sun8i' in '../linux-image-dev-sun8i_5.16_armhf.deb'.
dpkg-genchanges: warning: package linux-libc-dev-dev-sun8i in control file but not in files list
dpkg-genchanges: binary-only upload (no source code included)
[ o.k. ] Creating board support package [ orangepipc dev ]
[ o.k. ] Fingerprinting [ Armbian 5.16 Orangepipc Debian jessie dev ]
[ o.k. ] Building package [ linux-jessie-root-dev-orangepipc ]
[ o.k. ] Starting build process for [ orangepipc jessie ]
mount: permission denied
[ o.k. ] Creating new rootfs for [ jessie ]
[ o.k. ] Installing base system [ Stage 1/2 ]
I: Retrieving InRelease 
I: Failed to retrieve InRelease
I: Retrieving Release 
E: Failed getting release file http://localhost:3142/httpredir.debian.org/debian/dists/jessie/Release
cp: cannot create regular file '/root/output/cache/sdcard/usr/bin/': No such file or directory
[ o.k. ] Installing base system [ Stage 2/2 ]
chroot: failed to run command '/bin/bash': No such file or directory
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:166 ]
[ error ] Debootstrap base system second stage failed 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:553 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

`# ./compile.sh BOARD=orangepipc PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=dev `

zador-blood-stained, 2016-07-04:
@kometchtech 
Please add `NO_APT_CACHER=yes` to your build options

kometchtech, 2016-07-04:
@zador-blood-stained 

Thanks for the reply.
You could advance the progress so far NO_APT_CACHER = yes, pointed out that the options granted.

However, which seems far less errors.

```
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
cp: cannot create regular file '/root/output/cache/sdcard/etc/lirc/lircd.conf': No such file or directory
[ o.k. ] Installing extra applications and drivers 
[ o.k. ] Merging and packaging linux firmware [ @host ]
[ o.k. ] Installing linux firmware [ 5.16 ]
dpkg: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
 missing description
dpkg: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
 missing architecture
dpkg: error processing archive /tmp/armbian-firmware_5.16_all.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 /tmp/armbian-firmware_5.16_all.deb
[ o.k. ] Building deb [ hostapd ]
[ o.k. ] Downloading sources [ hostapd ]
[ o.k. ] Compiling hostapd [ v2.5 ]
[ error ] ERROR in function compiling [ hostapd.sh:72 ]
[ error ] Error building [ hostapd ]
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:553 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

PostScript:
Cannot add GUI as a permanent setting for this option?

zador-blood-stained, 2016-07-04:
> sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory

Can I see full build log? This suggests that there were errors before that line.

> Cannot add GUI as a permanent setting for this option?

I'll adjust some things later. `NO_APT_CACHER` will be set to `yes` by default if container environment is detected and `NO_APT_CACHER=no` is not specified explicitly (to allow using apt-cacher on host).

kometchtech, 2016-07-04:
@zador-blood-stained 

> sed: can't read /root/output/cache/sdcard/etc/lirc/hardware.conf: No such file or directory
> 
> Can I see full build log? This suggests that there were errors before that line.

make sure you get a log so.
[logs-04_07_2016-19_40_12.zip](https://github.com/igorpecovnik/lib/files/346798/logs-04_07_2016-19_40_12.zip)

zador-blood-stained, 2016-07-18:
I pushed some container-related fixes, please test and report if building process works better now.

lenormf, 2016-09-04:
Hi,

I've been having the same issues building an image within a docker container, here's what I've tried:

```
./compile.sh BOARD=orangepipcplus PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=default CLEAN_LEVEL=cache LIB_TAG=docker-test
```

```
[ o.k. ] Creating blank image for rootfs [ 1189 MiB ]
1.16GB [ 300MB/s] [=====================================================================================================================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
mkfs.ext4: No such device or address while trying to determine filesystem size
tune2fs: No such device or address while trying to open /dev/loop0p1
Couldn't find valid filesystem superblock.
mount: /dev/loop0p1 is not a valid block device
[ o.k. ] Copying files to image [ tmprootfs.raw ]
        688.46M  99%  179.63MB/s    0:00:03 (xfr#37457, to-chk=0/47403)   

sent 691.25M bytes  received 754.29K bytes  197.72M bytes/sec
total size is 692.33M  speedup is 1.00
[ o.k. ] Copying files to /boot partition [ tmprootfs.raw ]
         14.56M  99%  256.64MB/s    0:00:00 (xfr#43, to-chk=0/47) 

sent 14.57M bytes  received 848 bytes  29.14M bytes/sec
total size is 14.56M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
grep:  | tee -a $DEST/debug/debootstrap.log: No such file or directory
[ o.k. ] Writing bootloader [ /dev/loop0 ]
umount: /root/armbian/output/cache/mount: not mounted
[ o.k. ] Done building [ /root/armbian/output/images/Armbian_5.17_Orangepipcplus_Debian_jessie_3.4.112.raw ]
[ o.k. ] Runtime [ 27 min ]
```

The same error with `/dev/loop0` occurs without any `LIB_TAG`.

zador-blood-stained, 2016-09-06:
@lenormf 
`docker-test` branch was merged to `master`, and it had fixes related to Xenial build environments.

Please try creating a container with Xenial x64 system inside, try out building and provide full build log and terminal output, I need to be sure that container detection properly works.

lenormf, 2016-09-07:
The build on a fresh `xenial` container worked flawlessly, thank you.

I've uploaded the script's output[1] and the actual debug log[2], after checking that the image is working properly on an SD card (which it does).

I used `compile.sh` with the following arguments:
```
./compile.sh BOARD=orangepipcplus PROGRESS_DISPLAY=plain RELEASE=jessie PROGRESS_LOG_TO_FILE=yes KERNEL_ONLY=no BUILD_DESKTOP=no BRANCH=default CLEAN_LEVEL=cache
```

I haven't tried anything else (kernel version, `trusty` compiler etc).

[1]: https://my.mixtape.moe/tvtjim.log
[2]: https://my.mixtape.moe/eijney.tar.xz

zador-blood-stained, 2016-09-07:
> I've uploaded the script's output and the actual debug log, after checking that the image is working properly on an SD card (which it does).

Thanks, there are still some  issues with installation of additional packages, and now I am aware of this.

zador-blood-stained, 2016-09-22:
Closing since base functionality works based on reports. Documentation should be updated to use Xenial build host inside containers.

closed 2016-09-22 by zador-blood-stained

amazingfate, 2022-05-09:
> @pietrushnic
> 
> I advise against doing image builds in docker. Docker containers are fine for doing armbian kernel builds, but problematic for image builds.
> 
> The armbian image build process relies on creating loopback image mounts. I spent a day and a half troubleshooting the process... Ultimately getting the loopback kernel modules to create proper loopback devices within the container repeatedly proved to be unreliable.
> 
> (I was using CentOS7 as my docker host with a ubuntu 14.04 Docker container for the build)
> 
> Your error is very familar to me. If you take a look at your container for another share you'll probably see that the sdcard folder is not mounted to a loopback to device.

I also meet your error with centos7 host. I reslove it my updating kernel to 5.4 from elrepo.