armbian/build #388: udevd: specified group 'sunxi-fel' unknown at startup

lauhub, 2016-07-01:
On A20-Lime2-emmc, I get the following message on each boot:

`udevd[242]: specified group 'sunxi-fel' unknown`

It appears twice:
```
udevd[246]: starting version 175
udevd[246]: specified group 'sunxi-fel' unknown

. ok 
udevd[246]: specified group 'sunxi-fel' unknown

```
Is it safe to remove the rule that cause this message (the file /lib/udev/rules.d/60-sunxi-tools.rules) ?
Or should one add the sunxi-fel group to the system ? In this last case, which users should be added to this group ?

zador-blood-stained, 2016-07-10:
> Is it safe to remove the rule that cause this message (the file /lib/udev/rules.d/60-sunxi-tools.rules) ?

Yes

> Or should one add the sunxi-fel group to the system ? In this last case, which users should be added to this group ?

This is the system group for users that should be able to access connected sunxi boards in FEL mode

Currently this group is not created automatically, this will be fixed in the future.

zador-blood-stained, 2016-08-19:
Since new release will contain separate sunxi-tools package with proper maintainer scripts, this is fixed.

closed 2016-08-19 by zador-blood-stained