armbian/build #390: Bulgaria - kernel testing, try hooking to kernelci.org, docs

lanefu, 2016-07-01:
Task originated from http://forum.armbian.com/index.php/topic/1325-claim-a-task-set-due-date-and-start-doing-it/

_naibmra - Bulgaria - kernel testing, try hooking to kernelci.org, docs - mid July_

lanefu, 2016-07-01:
Please keep all discussion for this issue on the forum topic available below:

[http://forum.armbian.com/index.php/topic/1530-390-kernel-testing-try-hooking-to-kernelciorg-docs/](http://forum.armbian.com/index.php/topic/1530-390-kernel-testing-try-hooking-to-kernelciorg-docs/)

zador-blood-stained, 2017-04-26:
Closing since we are running nightly builds which may be considered as kernel compilation testing for the most popular configurations.

closed 2017-04-26 by zador-blood-stained