armbian/build #391: Olimex A20 : Battery cannot be monitored

LittleScarabee, 2016-07-01:
Hi

I recently buy an Olimex A20 Lime2 4Gb.
I tested the last release "Xenial Server 5.14" and I still not be able to monitor my battery (8000 mah)
I've got this information on startup :
`
root@lime2:~# dmesg | grep axp
[    4.721639] axp20x-i2c 0-0034: AXP20x variant AXP209 found
[    4.740780] input: axp20x-pek as /devices/platform/soc@01c00000/1c2ac00.i2c/i2c-0/0-0034/axp20x-pek/input/input0
[    4.752024] ldo1: supplied by axp-ipsout
[    4.757262] ldo2: supplied by axp-ipsout
[    4.761971] ldo3: supplied by axp-ipsout
[    4.766239] ldo4: supplied by axp-ipsout
[    4.771651] dcdc2: supplied by axp-ipsout
[    4.776501] dcdc3: supplied by axp-ipsout
[    4.782804] axp20x-i2c 0-0034: Battery detection is disabled, enabling
[    4.790129] axp20x-i2c 0-0034: Backup (RTC) battery charging is disabled
[    4.797090] axp20x-i2c 0-0034: AXP20X driver loaded
`

See below result for all commands : 
`
root@LUIGI:~# cat /sys/power/axp_pmu/battery/amperage 
0
root@LUIGI:~# cat /sys/power/axp_pmu/battery/capacity 
0
root@LUIGI:~# cat /sys/power/axp_pmu/battery/charge
0
root@LUIGI:~# cat /sys/power/axp_pmu/battery/charging 
0
root@LUIGI:~# cat /sys/power/axp_pmu/battery/connected 
1
root@LUIGI:~# cat /sys/power/axp_pmu/battery/power 
0
root@LUIGI:~# cat /sys/power/axp_pmu/battery/ts_voltage 
0
root@LUIGI:~# cat /sys/power/axp_pmu/battery/voltage 
0
`

When my battery is connected and I use a power supply 5V / 1,5A : the battery doesn't go in charging phase. When my battery is connected and I user power supply 5V / 2A via USB OTG : the battery doesn't go in charging phase

Do you have any idea ?
Thanks
Cheers

zador-blood-stained, 2016-07-27:
Battery monitoring works OK for me on Cubietruck with mainline kernel. Are you sure that battery is properly connected? Does the board run from battery if you disconnect all other power sources?

ThomasKaiser, 2016-07-27:
> Do you have any idea?

Must be a connection problem since this works flawlessly with Lime2 (see [post 2](http://forum.armbian.com/index.php/topic/1631-tutorial-marriage-between-a20-and-h3-ups-mode-sunxi-pio-utility/))

LittleScarabee, 2016-07-27:
After check with the support of Olimex it's means I have got a problem on the power supply : because when I used my Olimex with power supply I cannot charge my battery. I'll keep you informed after check done by the support to confirm it's a hardware problem and not kernel.

closed 2016-09-10 by zador-blood-stained