armbian/build #392: documentation rework

lanefu, 2016-07-01:
Kriston - USA - documentation rework - July, August

Task from http://forum.armbian.com/index.php/topic/1325-claim-a-task-set-due-date-and-start-doing-it/page-4#entry11910

Current Dialog on this thread

http://forum.armbian.com/index.php/topic/1337-documentation-software-proposal-for-armbian-wiki/page-3#entry11902

Need to rename thread to link to issue

lanefu, 2016-07-01:
Please keep all discussion for this issue on the forum topic available below:

[http://forum.armbian.com/index.php/topic/1532-392-documentation-rework/](http://forum.armbian.com/index.php/topic/1532-392-documentation-rework/)

kriston, 2016-07-03:
I gladly claim this issue.
Forgive me if I don't do the GitHub way of claiming issues, just put me as an Assignee. (Sorry, I'm not sure if there is an option to claim an issue at the moment).

zador-blood-stained, 2017-01-03:
Documentation rework tasks are obsolete at this point.

closed 2017-01-03 by zador-blood-stained