armbian/build #397: maintain/fix DTS entries for some devices such I2C/SPI/W1

lanefu, 2016-07-01:
martinayotte - Canada - maintain/fix DTS entries for some devices such I2C/SPI/W1 - ASAP

Task from thread http://forum.armbian.com/index.php/topic/1325-claim-a-task-set-due-date-and-start-doing-it/?p=11253

lanefu, 2016-07-01:
Please keep all discussion for this issue on the forum topic available below:

[http://forum.armbian.com/index.php/topic/1537-397-maintainfix-dts-entries-for-some-devices-such-i2cspiw1/](http://forum.armbian.com/index.php/topic/1537-397-maintainfix-dts-entries-for-some-devices-such-i2cspiw1/)

martinayotte, 2016-07-23:
PR is now sent : https://github.com/igorpecovnik/lib/pull/421

I've faced some issues because of switch from 4.6.2 to 4.6.4 as well as u-boot from v2016-05 to v2106-07, but I get thru those ...

lanefu, 2016-10-21:
@martinayotte hey I realized your branch was merged a while back.. any objections to closing this github issue?

closed 2017-02-02 by zador-blood-stained