armbian/build #399: Prepare HOWTOs and package armbian-gc2035-fswebcam package

lanefu, 2016-07-01:
Gravelrash - UK - Prepare HOWTO's & package "armbian-gc2035-fswebcam package" - June

Task from thread http://forum.armbian.com/index.php/topic/1325-claim-a-task-set-due-date-and-start-doing-it/?p=10298

lanefu, 2016-07-01:
Please keep all discussion for this issue on the forum topic available below:

[http://forum.armbian.com/index.php/topic/1539-399-prepare-howtos-and-package-armbian-gc2035-fswebcam-package/](http://forum.armbian.com/index.php/topic/1539-399-prepare-howtos-and-package-armbian-gc2035-fswebcam-package/)

gravelrash, 2016-08-11:
https://github.com/gravelrash/lib/blob/master/extras/fswebcam.sh

zador-blood-stained, 2017-08-01:
fswebcam was aleady packaged for quite some time.

closed 2017-08-01 by zador-blood-stained