armbian/build #403: [can wait] upgrade of Odroid XU4 next kernel breaks "/boot" partition

golfromeo-fr, 2016-07-06:
A/
When creating a full image of Jessie Odroid XU4 next, everything is fine (still not /etc/fstab but ok!)

B/
When generating a kernel update (all the *.deb files) of Odroid-XU4 next, doing 
dpkg -i *.deb

1/ new dtb files are installed on partition #1 /boot

2/ partition #1 (/boot) is unmounted

3/ new zImage, config, uInitrd are installed directly on root partition in folder "/boot"

so a manual fix is needed, so Armbian is running fine again.

golfromeo-fr, 2016-09-21:
as reported by other members, uboot 2016.05-dirty seems to be working along the Hardkernel blobs

https://github.com/c0d3z3r0/odroid-xu4

so we could test and possibly get rid of dual boot partition for XU4

zador-blood-stained, 2016-09-21:
> so we could test and possibly get rid of dual boot partition for XU4

Please define "we" 😄 

As for separate fat boot partition - it can be removed by patching u-boot as long as it supports loading files from ext4, for example [pine64 default](https://github.com/igorpecovnik/lib/blob/master/patch/u-boot/u-boot-pine64-default/default-env.patch) now uses single ext4 partition scheme

golfromeo-fr, 2016-09-21:
@zador-blood-stained 
My friend Vlad would says "We" the Sovereign Russian People.
Here, more soberly, I only consider (without vodka) "We" as the people interested in having a nice Armbian image for XU4.

yes, right thank for the reminder "fatload" => "ext4load", the scratchy part is how C0d3z3r0 mixing the blobs with uboot 2016.05

zador-blood-stained, 2016-09-21:
Where is the facepalm smile... Hm. There isn't one. Strange, surely you need one when it's about software development. OK, will go the hard way: 👋 ➡️ 🙍 

Seriously - I mean we need someone who owns this board to test which commands are available in u-boot and dump default environment (for a start).

golfromeo-fr, 2016-09-21:
@zador-blood-stained 
;)
ok, let me try this

golfromeo-fr, 2016-09-22:
@zador-blood-stained 
http://forum.armbian.com/index.php/topic/1450-device-specific-odroid-xu4-part2-armbian-next/?p=16155

golfromeo-fr, 2016-10-01:
when #480 is pushed in "master", as XU4/next has single partition boot for SDcard & eMMC, probably we can close this one. thanks!

zador-blood-stained, 2016-10-01:
OK, we don't need multiple issues for the same problem anyway

closed 2016-10-01 by zador-blood-stained