armbian/build #414: armbianmonitor Banana Pi mainline not reading board id

LarsNorgaard, 2016-07-16:
The ArmbianMonitor script doesn't seem to get the board name for the banana pi.
No files are created in /etc/armbianmonitor/datasources for pmu.

If I manually run:
```
ln -fs /sys/power/axp_pmu/ac/voltage /etc/armbianmonitor/datasources/ac_voltage
		ln -fs /sys/power/axp_pmu/ac/amperage /etc/armbianmonitor/datasources/ac_current
		ln -fs /sys/power/axp_pmu/vbus/voltage /etc/armbianmonitor/datasources/usb_voltage
		ln -fs /sys/power/axp_pmu/vbus/amperage /etc/armbianmonitor/datasources/usb_current
		ln -fs /sys/power/axp_pmu/battery/voltage /etc/armbianmonitor/datasources/battery_voltage
		ln -fs /sys/power/axp_pmu/battery/amperage /etc/armbianmonitor/datasources/battery_current
		ln -fs /sys/power/axp_pmu/pmu/temp /etc/armbianmonitor/datasources/pmictemp
		ln -fs /sys/power/axp_pmu/battery/capacity /etc/armbianmonitor/datasources/battery_percent
		ln -fs /sys/power/axp_pmu/battery/charging /etc/armbianmonitor/datasources/battery_charging
		ln -fs /sys/power/axp_pmu/charger/amperage /etc/armbianmonitor/datasources/charger_current
		ln -fs /sys/power/axp_pmu/battery/connected /etc/armbianmonitor/datasources/battery_connected
		ln -fs /sys/power/axp_pmu/battery/charge /etc/armbianmonitor/datasources/battery_charge
```
Then I get PMU info like I should.
[image]

igorpecovnik, 2016-07-17:
It's not just banana but all A20 boards with mainline kernel. PMU driver is still WIP, so no one added this to monitor.

LarsNorgaard, 2016-07-17:
Ah I see. Still love all the work Zador, Kaiser and you Igor put into this project. :)

ThomasKaiser, 2016-07-17:
The whole `armbianmonitor-daemon` stuff is WiP. I worked on this maybe half an year ago but then decided to postpone this to wait in which direction development goes.

Now we have `/etc/armbian-release` to rely on and maybe we will create the various symlinks to datasources already at image creation time. The goal is to maintain a set of basic templates per $LINUXFAMILY (some families being treated identical since sun4i, sun5i and sun7i share all the interesting stuff due to all 3 different SoCs being accompanied by the same AXP209 PMU) and to add further stuff on demand and/or automagically. 

1-Wire thermal sensors for example. Every sensor that appears below `/sys/bus/w1/devices/` should be added automatically to `/etc/armbianmonitor/datasources` as symlink. Found symlinks should also automatically add to the template stuff. So by default a sensor available as `/sys/devices/w1_bus_master1/28-0000064cfef9` will be linked to `/etc/armbianmonitor/datasources/w1-28-0000064cfef9` and appears as `28-0000064cfef9` thermal source in RPi-Monitor's interface. But as soon as the user renames the symlink to eg. `/etc/armbianmonitor/datasources/w1-Living\ Room` then the temperature value will be shown as "Living Room" in the UI.

Next topic: Enabling SNMP and making all datasources below  `/etc/armbianmonitor/datasources` available through automatically adjusted `net-snmp` config. But this will take some time.

commit f1f8be4c (2016-10-25): Add some changed board names to armbianmonitor
Closes #414
closed 2016-10-25 by zador-blood-stained