armbian/build #418: Odroid C2 dist-upgrade from 3.14.70 to 3.14.73

gklipel, 2016-07-20:
Hi!

After last upgrade, from kernel 3.14.70 to 3.14.73, the board won't boot up unless the dtb file get replaced for the correspondant on 3.14.65+. Even replacing with dtb on 3.14.72 image, won't boot either.

It seems that this upgrade runs fine on SD Cards, but on EMMC it wont work. The board was running fine, but now it hangs every day.

What could cause this behavior?

igorpecovnik, 2016-07-29:
I made a manual upgrade (dpkg -i) with packages from our latest build source and upgrade from 3.14.72, which is currently latest for download, to 3.14.74

System was on eMMC and upgrade went fine.

zador-blood-stained, 2016-07-29:
Upgrade problem **can** be reproduced if downloaded prebuilt image (5.14) is upgraded via apt-get, but I have no idea why this happens and especially why this happens only on eMMC.

gklipel, 2016-07-29:
Thats right zador, it happens with 5.14 upgraded, only with eMMC.

Igor, yes, with fresh 5.16 it upgrades fine it seems... It happens from 5.14 to upper versions, only with eMMC

gklipel, 2016-07-29:
Could it be uboot?

zador-blood-stained, 2016-09-16:
Feel free to comment or reopen if latest release (5.20) didn't fix the issue.

closed 2016-09-16 by zador-blood-stained