armbian/build #422: FEL boot patches I needed to make in order to successfully boot

md2000, 2016-07-23:
FEL boot failed with an error like "Error: unrecognized/unsupported machine ID (r1 = 0x00000000)." 
I had to add
setenv machid 0x00001029
to userpatches/fel-boot.cmd to fix it. Does that happen to everyone?

Also, I had 2 IP addresses on different network interfaces. In fel-load.sh, the line
`FEL_LOCAL_IP=$(ifconfig $FEL_NET_IFNAME | sed -En 's/127.0.0.1//;s/.*inet (addr:)?(([0-9]*\.){3}[0-9]*).*/\2/p')`
then returns 2 lines of text, which causes the sed replace to fail, and the line "setenv nfs_ip FEL_LOCAL_IP" remains in boot.cmd. I added "| tail -n 1" locally to fix it, which happens to select the correct IP in my case (wlan0 instead of eth0). This might be a rare enough case to not bother?

zador-blood-stained, 2016-07-23:
> I had to add
setenv machid 0x00001029
to userpatches/fel-boot.cmd to fix it. Does that happen to everyone?

Yes, this is limitation of kernel/u-boot combination for sun8i LINUXFAMILY. A better (automated) solution exists for dealing with this case: https://github.com/igorpecovnik/lib/blob/master/scripts/fel-hooks.sh.example

> Also, I had 2 IP addresses on different network interfaces.

This is exactly why FEL_NET_IFNAME was added, for example you can set `FEL_NET_IFNAME=eth0` in userpatches/lib.config

md2000, 2016-07-23:
Thanks. *smack my head*

Both of these solutions are even described in the documentation. I'd glossed over them while trying to set up FEL, and didn't think to look back when troubleshooting.

closed 2016-07-23 by md2000