armbian/build #424: Some packages missing in /lib/firmware

martinayotte, 2016-07-24:
I've discovered that some firmwares are now missing in /lib/firmware, such as the realtek.
They were present in older builds, but they are not in the recent 4.6.4.

zador-blood-stained, 2016-07-25:
What files exactly are missing? If this firmware is available in official Debian/Ubuntu packages, then it is normal that it's not present by default.

martinayotte, 2016-07-25:
Yes, they are firmware-realtek and firmware-ralink.
But they were pre-installed before until recently directly in the built images.
It is because it is difficult to install them without having network connectivity.

igorpecovnik, 2016-07-29:
I double check the diff but we have all firmware files from those two packages except: https://github.com/igorpecovnik/lib/tree/master/bin/firmware-overlay/RTL8192SU
added and booted latest Opi PC+ from last build and wireless works fine. Closing

closed 2016-07-29 by igorpecovnik

martinayotte, 2016-07-29:
Ok for the RTL8192SU, but I was also talking about the whole /lib/firmware/rtlwifi that was present in older builds but not in the recent ones.

igorpecovnik, 2016-07-29:
We used to have one zip with firmware(s) but missing ones are included in overlay. I can't find what could be missing. Of course I am talking this for latest image build from sources. We had some missing in between and on upgraded images problems could exists.

martinayotte, 2016-07-29:
Yes, I'm always preparing my images from sources. I though that maybe some kind of "apt-get install firmware-realtek" were present before and that it has been removed recently, but I wasn't to find such changes in the history.

zador-blood-stained, 2016-07-29:
The only related change I remember was disabling installation of firmware package built with kernel: https://github.com/igorpecovnik/lib/blob/master/distributions.sh#L104-L108

martinayotte, 2016-07-29:
Thanks ! That is probably that, I will try to see if enabling it back fix it...
Do you know the reasons why it have been commented out ?

igorpecovnik, 2016-07-30:
The reason was to move thing under one hood, aka "armbian-firmware" and since there are duplicates, one must went out.