armbian/build #426: hdmi_edid.c patch path is different.

trustfarm-dev, 2016-07-25:
hi,,

--- a/drivers/video/sunxi/hdmi/aw/hdmi_edid.c
+++ b/drivers/video/sunxi/hdmi/aw/hdmi_edid.c

the file is not compiled in orangepi kernel building.
It should be changed to  
---a/drivers/video/sunxi/hdmi2/aw/hdmi_edid.c
+++ b/drivers/video/sunxi/hdmi2/aw/hdmi_edid.c

jernejsk, 2016-10-16:
which patch are you talking about?

zador-blood-stained, 2016-10-16:
I would guess this: https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-default/0026-h3-hdmi-additional-mode-800x480p-from-lobodis-initia.patch

jernejsk, 2016-10-16:
Well, that is certainly an error, but hopefully it won't matter anymore. I'm very close to implement resolution setting by EDID.

zador-blood-stained, 2016-12-28:
Closing since EDID rework for the legacy kernel was abandoned and in general it is an "upstream" problemof this project - we don't have resources for kernel modifications as large as this.

closed 2016-12-28 by zador-blood-stained