armbian/build #427: [Cubietruck] next building error

kometchtech, 2016-07-25:
Hi,

Determine which behavior in Cubietruck and tried the next build fails with error.
For more information check the log attached.

[compilation.zip](https://github.com/igorpecovnik/lib/files/381641/compilation.zip)

zador-blood-stained, 2016-07-25:
This was fixed by adding linaro 5.3.1 compiler and compilation works fine for me, so we need more reports to confirm that something still doesn't work.

kometchtech, 2016-07-25:
Hi, @zador-blood-stained 

Last it will stop with the following message.

```
  LD      common/built-in.o
  LD      cmd/built-in.o
[ error ] ERROR in function compile_uboot [ common.sh:94 ]
[ error ] U-boot file not found [ u-boot-sunxi-with-spl.bin ]
[ o.k. ] Process terminated 
```

igorpecovnik, 2016-07-25:
Works fine here too. I made several compilations today.

zador-blood-stained, 2016-07-25:
@kometchtech 
Maybe your toolchain download was interrupted, I'll try to make this process more reliable in the future, Please delete 5.3 armhf toolchain from toolchains subdirectory and try again.

kometchtech, 2016-07-25:
@zador-blood-stained 

I tried again and removed, but will stop with an error in the use of the following.
```
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config.h
  GEN     include/autoconf.mk
  GEN     include/autoconf.mk.dep
  GEN     spl/include/autoconf.mk
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  CFG     u-boot.cfg
  CHK     include/generated/version_autogenerated.h
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  HOSTCC  tools/bmp_logo
  HOSTCC  tools/gen_eth_addr
  HOSTCC  tools/img2srec
  HOSTCC  tools/mkenvimage.o
  HOSTCC  tools/os_support.o
  WRAP    tools/lib/crc32.c
  HOSTCC  tools/aisimage.o
  HOSTCC  tools/atmelimage.o
  WRAP    tools/common/bootm.c
  HOSTCC  tools/default_image.o
  WRAP    tools/lib/fdtdec_common.c
  WRAP    tools/lib/fdtdec.c
  HOSTCC  tools/fit_common.o
  HOSTCC  tools/fit_image.o
  WRAP    tools/common/image-fit.c
  HOSTCC  tools/image-host.o
  WRAP    tools/common/image.c
  LDS     u-boot.lds
  HOSTCC  tools/imagetool.o
  HOSTCC  tools/imximage.o
  HOSTCC  tools/kwbimage.o
  WRAP    tools/lib/md5.c
  HOSTCC  tools/lpc32xximage.o
  HOSTCC  tools/mxsimage.o
  HOSTCC  tools/omapimage.o
  HOSTCC  tools/pblimage.o
  HOSTCC  tools/pbl_crc32.o
  WRAP    tools/lib/rc4.c
  HOSTCC  tools/rkcommon.o
  HOSTCC  tools/rkimage.o
  HOSTCC  tools/rksd.o
  HOSTCC  tools/rkspi.o
  HOSTCC  tools/socfpgaimage.o
  WRAP    tools/lib/sha1.c
  WRAP    tools/lib/sha256.c
  WRAP    tools/common/hash.c
  HOSTCC  tools/ublimage.o
  HOSTCC  tools/zynqimage.o
  HOSTCC  tools/zynqmpimage.o
  WRAP    tools/lib/libfdt/fdt.c
  WRAP    tools/lib/libfdt/fdt_ro.c
  WRAP    tools/lib/libfdt/fdt_rw.c
  WRAP    tools/lib/libfdt/fdt_strerror.c
  WRAP    tools/lib/libfdt/fdt_wip.c
  WRAP    tools/lib/libfdt/fdt_region.c
[... 235 lines cut ...]
  CC      common/cli_readline.o
  CC      cmd/mmc.o
  CC      common/command.o
  CC      cmd/net.o
  CC      cmd/part.o
  CC      common/s_record.o
  CC      cmd/pcmcia.o
  CC      cmd/pxe.o
  CC      cmd/scsi.o
  CC      cmd/setexpr.o
  CC      cmd/test.o
  CC      cmd/usb.o
  CC      cmd/fastboot.o
  CC      cmd/usb_mass_storage.o
  CC      common/xyzModem.o
  CC      cmd/ximg.o
  CC      cmd/dfu.o
  CC      cmd/nvedit.o
  CC      cmd/disk.o
  LD      common/built-in.o
  LD      cmd/built-in.o
[ error ] ERROR in function compile_uboot [ common.sh:94 ]
[ error ] U-boot file not found [ u-boot-sunxi-with-spl.bin ]
[ o.k. ] Process terminated 
```

igorpecovnik, 2016-07-26:
Are you using any custom patches?

kometchtech, 2016-07-26:
@igorpecovnik 

Fails if the container you created in the following environments.
Was tried, "14.04 LTS container, no problem.

```
# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04 LTS
Release:        16.04
Codename:       xenial
```

kometchtech, 2016-07-28:
I looked, but there was no corresponding file.

```
root@e00b07e58174:~/sources/u-boot/v2016.07/arch/arm/cpu/armv7/sunxi# ls -l
total 60
-rw-r--r-- 1 root root   472 Jul 26 08:27 Makefile
-rw-r--r-- 1 root root   126 Jul 26 08:27 config.mk
-rw-r--r-- 1 root root   951 Jul 26 08:27 fel_utils.S
-rw-r--r-- 1 root root  5996 Jul 26 08:27 psci.c
-rw-r--r-- 1 root root   223 Jul 26 08:30 psci.su
-rw-r--r-- 1 root root  1793 Jul 26 08:27 psci_head.S
-rw-r--r-- 1 root root  1632 Jul 26 08:30 psci_head.o
-rw-r--r-- 1 root root    70 Jul 26 08:27 start.c
-rw-r--r-- 1 root root  2635 Jul 26 08:27 timer.c
-rw-r--r-- 1 root root 10672 Jul 26 08:30 timer.o
-rw-r--r-- 1 root root   704 Jul 26 08:27 tzpc.c
-rw-r--r-- 1 root root  1149 Jul 26 08:27 u-boot-spl.lds
```

zador-blood-stained, 2016-07-28:
Their names start with a dot, so they are "hidden"
```
➜  armbian  % ls -l sources/u-boot/v2016.07/arch/arm/cpu/armv7/sunxi
total 116
-rw-rw-r-- 1 root root 24804 Jul 27 20:38 built-in.o
-rw-r--r-- 1 root root   126 Jul 12 00:25 config.mk
-rw-r--r-- 1 root root   951 Jul 12 00:25 fel_utils.S
-rw-r--r-- 1 root root   472 Jul 12 00:25 Makefile
-rw-r--r-- 1 root root  5996 Jul 12 00:25 psci.c
-rw-rw-r-- 1 root root  1664 Jul 27 20:38 psci_head.o
-rw-r--r-- 1 root root  1793 Jul 12 00:25 psci_head.S
-rw-rw-r-- 1 root root 13416 Jul 27 20:38 psci.o
-rw-rw-r-- 1 root root   224 Jul 27 20:38 psci.su
-rw-r--r-- 1 root root    70 Jul 12 00:25 start.c
-rw-r--r-- 1 root root  2635 Jul 12 00:25 timer.c
-rw-rw-r-- 1 root root 10212 Jul 27 20:38 timer.o
-rw-rw-r-- 1 root root   202 Jul 27 20:38 timer.su
-rw-r--r-- 1 root root   704 Jul 12 00:25 tzpc.c
-rw-rw-r-- 1 root root  3524 Jul 27 20:38 tzpc.o
-rw-rw-r-- 1 root root    31 Jul 27 20:38 tzpc.su
-rw-r--r-- 1 root root  1149 Jul 12 00:25 u-boot-spl.lds
➜  armbian  % ls -la sources/u-boot/v2016.07/arch/arm/cpu/armv7/sunxi
total 308
drwxr-xr-x  2 root root  4096 Jul 27 20:38 .
drwxr-xr-x 22 root root  4096 Jul 27 20:38 ..
-rw-rw-r--  1 root root 24804 Jul 27 20:38 built-in.o
-rw-r--r--  1 root root   259 Jul 27 20:38 .built-in.o.cmd
-rw-r--r--  1 root root   126 Jul 12 00:25 config.mk
-rw-r--r--  1 root root   951 Jul 12 00:25 fel_utils.S
-rw-r--r--  1 root root   472 Jul 12 00:25 Makefile
-rw-r--r--  1 root root  5996 Jul 12 00:25 psci.c
-rw-rw-r--  1 root root  1664 Jul 27 20:38 psci_head.o
-rw-r--r--  1 root root 14676 Jul 27 20:38 .psci_head.o.cmd
-rw-r--r--  1 root root  1793 Jul 12 00:25 psci_head.S
-rw-rw-r--  1 root root 13416 Jul 27 20:38 psci.o
-rw-r--r--  1 root root 80442 Jul 27 20:38 .psci.o.cmd
-rw-rw-r--  1 root root   224 Jul 27 20:38 psci.su
-rw-r--r--  1 root root    70 Jul 12 00:25 start.c
-rw-r--r--  1 root root  2635 Jul 12 00:25 timer.c
-rw-rw-r--  1 root root 10212 Jul 27 20:38 timer.o
-rw-r--r--  1 root root 80365 Jul 27 20:38 .timer.o.cmd
-rw-rw-r--  1 root root   202 Jul 27 20:38 timer.su
-rw-r--r--  1 root root   704 Jul 12 00:25 tzpc.c
-rw-rw-r--  1 root root  3524 Jul 27 20:38 tzpc.o
-rw-r--r--  1 root root  3182 Jul 27 20:38 .tzpc.o.cmd
-rw-rw-r--  1 root root    31 Jul 27 20:38 tzpc.su
-rw-r--r--  1 root root  1149 Jul 12 00:25 u-boot-spl.lds
➜  armbian  %

```

kometchtech, 2016-07-28:
Excuse me.

Attach because it was able to confirm the file.

```
root@e00b07e58174:~/sources/u-boot/v2016.07/arch/arm/cpu/armv7/sunxi# ls -la
total 172
drwxr-xr-x  2 root root  4096 Jul 28 12:14 .
drwxr-xr-x 22 root root  4096 Jul 28 12:14 ..
-rw-r--r--  1 root root  2487 Jul 28 12:14 .psci.o.d
-rw-r--r--  1 root root 14616 Jul 28 12:14 .psci_head.o.cmd
-rw-r--r--  1 root root 80083 Jul 28 12:14 .timer.o.cmd
-rw-r--r--  1 root root   472 Jul 28 12:03 Makefile
-rw-r--r--  1 root root   126 Jul 28 12:03 config.mk
-rw-r--r--  1 root root   951 Jul 28 12:03 fel_utils.S
-rw-r--r--  1 root root  5996 Jul 28 12:03 psci.c
-rw-r--r--  1 root root   223 Jul 28 12:14 psci.su
-rw-r--r--  1 root root  1793 Jul 28 12:03 psci_head.S
-rw-r--r--  1 root root  1632 Jul 28 12:14 psci_head.o
-rw-r--r--  1 root root    70 Jul 28 12:03 start.c
-rw-r--r--  1 root root  2635 Jul 28 12:03 timer.c
-rw-r--r--  1 root root 10672 Jul 28 12:14 timer.o
-rw-r--r--  1 root root   202 Jul 28 12:14 timer.su
-rw-r--r--  1 root root   704 Jul 28 12:03 tzpc.c
-rw-r--r--  1 root root  1149 Jul 28 12:03 u-boot-spl.lds
```
[cmd.zip](https://github.com/igorpecovnik/lib/files/388359/cmd.zip)

zador-blood-stained, 2016-07-28:
OK, I think I found the problem.
```
➜  armbian  % apt-cache policy gcc-5-arm-linux-gnueabihf
gcc-5-arm-linux-gnueabihf:
  Installed: 5.4.0-6ubuntu1~16.04.1cross1
  Candidate: 5.4.0-6ubuntu1~16.04.1cross1
  Version table:
 *** 5.4.0-6ubuntu1~16.04.1cross1 500
        500 http://ru.archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     5.3.1-14ubuntu2cross1 500
        500 http://ru.archive.ubuntu.com/ubuntu xenial/main amd64 Packages
```
So xenial repository provides GCC 5.3.1, and xenial-updates provides 5.4.0. So setting requirement to 5.2 should fix the issue: https://github.com/igorpecovnik/lib/commit/ba941c8c270fc6376d46b4a9d6ffe11444819614

kometchtech, 2016-07-29:
Alright now pass compilation.
[logs-29_07_2016-12_45_05.zip](https://github.com/igorpecovnik/lib/files/390554/logs-29_07_2016-12_45_05.zip)

closed 2016-07-29 by zador-blood-stained