armbian/build #429: Bug in general.sh download_toolchain()

issalig, 2016-07-28:
https://github.com/igorpecovnik/lib/blob/master/general.sh#L468

This line in function download_toolchain()
tar --overwrite -xf $filename && touch $dirname/.download-complete
sholud be
tar --overwrite -xf $filename && touch $SRC/toolchains/$dirname/.download-complete
otherwise always downloads toolchains

zador-blood-stained, 2016-07-28:
Strange. Works for me, but it won't hurt to use absolute path instead of relative.

closed 2016-07-28 by zador-blood-stained

issalig, 2016-07-28:
With relative will work only if .download-complete already existed but if it was a new download the touch in the non-existing path will not create .download-complete.

Anyway, it is fixed now. Thanks for the fast fix.

commit 1951b277 (2016-07-28): Fix possible bug in download_toolchain
Closes #429