armbian/build #430: RT Linux

globalkrishna, 2016-07-29:
any way of using the linux-linaro-lsk-v4.4-rt
[https://git.linaro.org/gitweb?p=kernel/linux-linaro-stable.git;a=shortlog;h=refs/heads/linux-linaro-lsk-v4.4-rt](url)

zador-blood-stained, 2016-07-29:
If you need RT kernel it may be easier to compile current version (4.6) with [RT patch](https://www.kernel.org/pub/linux/kernel/projects/rt/4.6/). You will need to unpack patch-4.6.4-rt8.patch.xz to userpatches/kernel/sunxi-next (I'm assuming you are talking about A10/A20 based board) and compile a kernel using standard procedure.

globalkrishna, 2016-07-29:
Thought of it.
Quoting from Linaro site:
"Linaro Stable Kernel (LSK) aims to provide a stable basis for system integration of ARM hardware, integrating Linaro developed features available in mainline with kernel.org Long Term Stable (LTS) releases to allow integration of those features in production systems."

Is it possible to provide an option in the make script to pull Kernel source from Linaro?

I intend to use Orange Pi-one/lite to run LED display boards.
I have done it with Beablebone Black using built in Programmable Real Time Unit (PRU).  
But, such a feature is not available in H3 devices.

igorpecovnik, 2016-07-29:
> Is it possible to provide an option in the make script to pull Kernel source from Linaro?

You will not be able to boot the board. Just use the kernel we provide and add functionality you need.

zador-blood-stained, 2016-07-29:
> I intend to use Orange Pi-one/lite to run LED display boards.

No proper support in mainline yet for H3, especially in such an old kernel (4.4). You can try using RT patch for 4.6 with sun8i-dev branch and see if it applies without any errors and compiles after that.

globalkrishna, 2016-07-29:
Thanks a lot for the all the information and support.

closed 2016-07-29 by globalkrishna