armbian/build #432: SS5 socks server can not work properly in armbian

1558204459, 2016-08-04:
SS5 can not work in armbian,but it works at debian and ubuntu system
I think it should be the system's problem,but i don't where the problem is.

igorpecovnik, 2016-08-04:
You mean it works good on x86 system? 

Are there any kernel dependencies and on which board you were trying to run this?

1558204459, 2016-08-04:
Yes it works on x86
I tried the cubian system in cubieboard,but it still not work properly
And when i compile the SS5 in armbian ubuntu,it can not be compiled
This the error message:
../include/SS5Server.h:35:3: warning: inline function ‘S5ChildClose’ declared but never defined
   S5ChildClose(  int exitcode,
   ^

gcc -g -O2 -DLINUX -D_FILE_OFFSET_BITS=64 -I . -I ../include      -c -o SS5Debug.o SS5Debug.c
gcc -g -O2 -DLINUX -D_FILE_OFFSET_BITS=64 -I . -I ../include      -c -o SS5GSSApi.o SS5GSSApi.c
gcc -g -O2 -DLINUX -D_FILE_OFFSET_BITS=64 -I . -I ../include      -c -o SS5Modules.o SS5Modules.c
In file included from SS5Modules.c:20:0:
../include/SS5Server.h:35:3: warning: inline function ‘S5ChildClose’ declared but never defined
   S5ChildClose(  int exitcode,
   ^

gcc -g -O2 -DLINUX -D_FILE_OFFSET_BITS=64 -I . -I ../include    -o ss5 SS5Main.o SS5Core.o SS5Server.o SS5Utils.o SS5Thread.o SS5Debug.o SS5GSSApi.o SS5Modules.o ../common/SS5OpenLdap.o ../common/SS5Radius.o ../common/SS5MySql.o -lpthread -rdynamic -ldl -lssl -lldap -lpam -lpam_misc -lcrypto
SS5Main.o: In function `main':
/root/tmp/ss5-3.8.9/src/SS5Main.c:339: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Main.c:346: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Main.c:323: undefined reference to `S5ChildClose'
SS5Core.o: In function `S5Core':
/root/tmp/ss5-3.8.9/src/SS5Core.c:175: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:419: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:623: undefined reference to `S5DebugUpstreamInfo'
/root/tmp/ss5-3.8.9/src/SS5Core.c:1458: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:419: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:158: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:160: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:167: undefined reference to `S5ChildClose'
SS5Core.o:/root/tmp/ss5-3.8.9/src/SS5Core.c:165: more undefined references to `S5ChildClose' follow
SS5Core.o: In function `S5Core':
/root/tmp/ss5-3.8.9/src/SS5Core.c:759: undefined reference to `S5DebugUpstreamInfo'
/root/tmp/ss5-3.8.9/src/SS5Core.c:792: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:816: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:317: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:839: undefined reference to `S5DebugUdpRequestInfo'
/root/tmp/ss5-3.8.9/src/SS5Core.c:842: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:843: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:866: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:397: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:465: undefined reference to `S5DebugFacilities'
/root/tmp/ss5-3.8.9/src/SS5Core.c:504: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:1206: undefined reference to `S5ChildClose'
/root/tmp/ss5-3.8.9/src/SS5Core.c:985: undefined reference to `S5ChildClose'
collect2: error: ld returned 1 exit status
Makefile:32: recipe for target 'ss5' failed
make[1]: *** [ss5] Error 1
make[1]: Leaving directory '/root/tmp/ss5-3.8.9/src'
Makefile:31: recipe for target 'src' failed
make: *** [src] Error 2

zador-blood-stained, 2016-08-04:
It may be related to compiler version, or it just wasn't designed to run on anything other than x86/x64

1558204459, 2016-08-04:
I will try another system in the cubieboard,Then I will give you the feedback

zador-blood-stained, 2016-08-18:
Compiling sources from ss5-3.8.9-8.tar.gz works in Jessie (after installing `libpam0g-dev` and `libldap2-dev`) but fails in Xenial due to incompatibility with [GCC 5](https://gcc.gnu.org/gcc-5/porting_to.html).

So check your compiler version and switch to older one if default doesn't work.

Closing since this is not Armbian related issue.

closed 2016-08-18 by zador-blood-stained

cjhxmx, 2020-12-09:
try this command:  ./configure  CFLAGS=-fgnu89-inline