armbian/build #434: Add a basic DRM driver for H3 for custom HDMI modes without kernel recompile

zldo, 2016-08-05:
I found a patch adding DRM driver for Allwinner H3 (Allwinner new display interface DE2) 
https://patchwork.kernel.org/patch/8191881/
Anybody can help integrate it into armbian?

vincent-legoll, 2016-10-13:
You can find the developer's web page [here](http://moinejf.free.fr/opi2/index.html), with the most up to date patch.

ThomasKaiser, 2016-10-24:
In case you want to try it out simply follow the [steps outlined here](https://github.com/igorpecovnik/lib/issues/508). It's just exchanging the branch in [lib/config/sources/sun8i.conf](https://github.com/igorpecovnik/lib/blob/master/config/sources/sun8i.conf#L21).

Please report back. Though this requires use of Armbian's build system (and no, if we decide not to use the `orange-pi-4.9-hdmi` branch now then there's no easy way to test this out -- but using the build system is pretty easy once you [meet the requirements](http://docs.armbian.com/Developer-Guide_Build-Preparation/))

zador-blood-stained, 2016-11-30:
Based on the [discussion](https://groups.google.com/forum/#!topic/linux-sunxi/96vcJ1kdXXw) this driver won't be merged to mainline soon. If it will be a part of current `sun8i-dev` branch that we use (i.e. [this one](https://github.com/megous/linux/commits/orange-pi-4.9), it will be available as part of sun8i-dev kernel.

closed 2016-11-30 by zador-blood-stained