armbian/build #435: ERROR in function create_rootfs_cache

multivac61, 2016-08-08:
I am building a .img file for the Allwinner M1 H3 based development board. I want to build the legacy kernel and use xenial. When I get to Debootstrap (stage 2/2)... the compile script stops and I get the following errors:

```
cp: cannot create regular file '/home/horigome/genki/output/cache/sdcard/usr/bin/': No such file or directory
cp: cannot stat '/usr/share/keyrings/debian-archive-keyring.gpg': No such file or directory
[ o.k. ] Installing base system [ Stage 2/2 ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
...
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:168 ]
[ error ] Debootstrap base system second stage failed 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:556 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

The host machine is running 64bit Ubuntu 16.04 LTS.

igorpecovnik, 2016-08-08:
Try to remove cache 
`output/cache/rootfs`
and try again. I just build and had no problems.

multivac61, 2016-08-08:
I tried starting from scratch and doing it under /tmp/ but I get the exact same error:

```
cp: cannot create regular file '/tmp/output/cache/sdcard/usr/bin/': No such file or directory
cp: cannot stat '/usr/share/keyrings/debian-archive-keyring.gpg': No such file or directory
[ o.k. ] Installing base system [ Stage 2/2 ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:168 ]
[ error ] Debootstrap base system second stage failed 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:556 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated
```

zador-blood-stained, 2016-08-08:
@multivac61
Please provide output of `systemctl status apt-cacher-ng -l`

multivac61, 2016-08-08:
I get 
```
● apt-cacher-ng.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)
```

zador-blood-stained, 2016-08-08:
Can you provide full output from compilation attempt? If ccache wasn't installed then there may be issues with other build dependencies.

multivac61, 2016-08-08:
Allright so I am past Installing base system and now it looks like the script is making the .img file. I have had a couple of problems with the compile scripts which I have sorted out with the following steps:

1. The PATH did not include arm-linux-gnueabihf-gcc so I had to add it.
2. I had to pass on the path when running .compile.sh as root, e.g. `sudo ./compile.sh PATH=$PATH..`
3. I had to manually install both apt-cacher-ng and debian-archive-keyring

multivac61, 2016-08-08:
Now I get a new error: (I don't know if this should be another issue):
```
[ o.k. ] Writing bootloader [ /dev/loop0 ]
[ error ] ERROR in function write_uboot [ common.sh:377 ]
[ error ] U-boot failed to install [ @host ]
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ debootstrap-ng.sh:556 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated
```

zador-blood-stained, 2016-08-08:
> The PATH did not include arm-linux-gnueabihf-gcc so I had to add it.

> [ error ] ERROR in function write_uboot [ common.sh:377 ]

Toolchains are downloaded and added to PATH automatically depending on build requirements. In case you added wrong toolchain version to PATH, you may have u-boot compilation errors and u-boot write errors as a consequence, though this should be detected earlier during build process.

igorpecovnik, 2016-08-08:
I just made a clean test inside virtual machine as stated in out "how-to". Creating Orange Pi PC desktop image.

No problems whatsoever.

multivac61, 2016-08-08:
Alright, I will try and run it inside a Vagrant virtual machine and then report.

multivac61, 2016-08-08:
Starting with a completely blank Ubuntu Xenial 16.04 LTS vagrant box and following the "how-to" I get the following errors when running .compile.sh:

```
[ error ] ERROR in function compile_kernel [ common.sh:182 ]
[ error ] Kernel was not built [ @host ]
```

The last lines of output.log are:

```
...
Displaying message: ... 02-0013-w1_h3.patch succeeded info
Displaying message: ... 02-0014-no_dev_console.patch succeeded info
Displaying message: ... 02-0015-openelec-audio-and-video-fixes.patch succeeded info
Displaying message: ... 02-rtl2832.patch succeeded info
Displaying message: ... add_8189es_v4.3.18.1_15373.20151005.patch succeeded info
Displaying message: ... add_8189fs_v4.3.24_15589.20151023.patch succeeded info
Displaying message: ... bash_to_afterinstall.patch succeeded info
Displaying message: ... cedar-remove-insecure-code.patch succeeded info
Displaying message: ... cpufreq-add-more-frequencies.patch succeeded info
Displaying message: ... enable_g_ether.patch succeeded info
Displaying message: ... fix-kswapd.patch succeeded info
Displaying message: ... linux-82-fix-build-on-non-amd64-hosts.patch succeeded info
Displaying message: ... linux-88-enable-tv.patch succeeded info
Displaying message: ... m2-plus-ap6212-wireless.patch succeeded info
Displaying message: ... packaging-3.4.x-DEFAULT-with-postinstall-scripts.patch succeeded info
Displaying message: ... remove-sunxi-debug.patch succeeded info
Displaying message: ... sun8i-led-behaviour-through-fex.patch succeeded info
Displaying message: ... sun8i-simple-cpu-corekeeper.patch succeeded info
Displaying message: ... tbs5520.patch succeeded info
Displaying message: ... wdt_reduce_logging.patch succeeded info
Displaying message: ... z-0003-add-additional-video-modes.patch succeeded info
Displaying message: Compiling default kernel 3.4.112 info
Displaying message: Compiler version arm-linux-gnueabihf-gcc 5.4.0 info
Displaying message: Using kernel config file lib/config/kernel/linux-sun8i-default.config info
Displaying message: ERROR in function compile_kernel common.sh:182 err
Displaying message: Kernel was not built @host err
Displaying message: Process terminated  info
```

multivac61, 2016-08-09:
@igorpecovnik You can close this particular issue, it was due to my particular host setup. Thanks for great support!

closed 2016-08-09 by zador-blood-stained

mgilge, 2019-01-17:
@multivac61 I am having the same issues. Can you descrive what was wrong in the setip of the host system? How did you rectify the setup?